PX4/PX4-Autopilot #4224: dronekit and modes from pymavlink

mrpollo, 2016-04-12:
I have received an increased number of requests to look into the PX4 integration in DroneKit, one of the problems that is bugging people is that we are facing a mode problem when connecting

```
`>>> Exception in message handler for HEARTBEAT
```
DroneKit is using the [`mode_mapping_bynumber` helper from **mavutil.py**](https://github.com/mavlink/mavlink/blob/master/pymavlink/mavutil.py#L1416-L1433) and we seem to be facing modes not defined there.

I don't have much much time to spend debugging this I'm looking for help from the PX4 community, there are plenty issues on our dronekit repo but this should be the official

https://github.com/dronekit/dronekit-python/issues/606

dagar, 2016-07-19:
The PX4 custom modes mavlink values are defined here: https://github.com/PX4/Firmware/blob/master/src/modules/commander/px4_custom_mode.h#L46

LorenzMeier, 2016-12-26:
@mrpollo Status?

dagar, 2017-04-11:
Wasn't this resolved? @sanderux

sanderux, 2017-04-11:
I added and got merged a PR in both dronekit and pymavlink but mavutil needs to release a new version on pip before this will be operational. Not sure what the status is there

dagar, 2017-04-11:
Any idea who takes care of that?

sanderux, 2017-04-11:
i think tridge

closed 2017-04-12 by dagar