PX4/PX4-Autopilot #4230: USB serial connection problems with Pixhawk

bkueng, 2016-04-13:
Hi

I have a problem connecting to the pixhawk (fmuv2), via USB cable (using `screen /dev/ttyACM0`, `minicom` or QGC, all the same behavior). When I connect the cable, then wait several seconds and try to connect, it succeeds to connect, but I don't get a nsh shell (no reaction when pressing enter, just a blank screen). However when I connect during the first few seconds, right after the big blue led goes on, everything works and I get a nsh shell. When I then exit `screen` and restart it, it still works.
FW uploading also works, but only when I do it within the first few seconds.

I tried with two different boards, both the same behavior. On another computer, they don't have this issue, only on mine, a Thinkpad T440p.

I also tested this with the PixFalcon and it does not have the same problem, it occured only with the pixhawk.

Has anyone experienced similar problems?

LorenzMeier, 2016-04-18:
Have you de-installed the modem manager? Have you tried a different USB cable?

bkueng, 2016-04-18:
yes & yes

tumbili, 2016-04-18:
@LorenzMeier It seems strange that it correlates with the time he waits until he connects after boot.

tops4u, 2016-04-20:
I have a similar Problem (Mac OS X, AUAV-X2 or Pixracer). When connecting via USB to Pixhawk after it has been powered by Battery very often the connection would not work (with no Error in QGC). Only when disconnecting Battery and USB and restarting only with USB the connction will work. No such Problem when using Serial Telemetry.

LorenzMeier, 2016-04-20:
Are you using El Capitan?

tops4u, 2016-04-20:
Yes I do. Happend with all Versions of QGC to date (from 2.9 onwards). With PX4 1.2 as well as current master.

LorenzMeier, 2016-05-01:
@tops4u Its an issue with El Capitan. Make sure to unplug, close QGC and then only plug back in. That will work around it until Apple fixes it.

closed 2016-05-01 by LorenzMeier