PX4/PX4-Autopilot #4232: bad bytes in log

georgehines, 2016-04-13:
`make px4fmu-v2_default upload`
completed successfully to a Pixhawk.

Executed
`sdlog2 stop`
`sdlog2 start -t -r 200 -e -b 16`
`sdlog2 stop`
at the nsh prompt

Moved the logfile to my computer and dumped with mavlogdump.py, gave
`Skipped 74 bad bytes in log (73, 78, 70) remaining=496769`
`Skipped 74 bad bytes in log (73, 78, 70) remaining=496769`
`1969-12-31 16:00:00.02: FMT {Type : 2, Length : 55, Name : ATT, Format : fffffffffffff, Columns : qw,qx,qy,qz,Roll,Pitch,Yaw,RollRate,PitchRate,YawRate,GX,GY,GZ}`
etc.

Opening in FlightPlot fails with "Invalid header at 1"

[log001.zip](https://github.com/PX4/Firmware/files/218102/log001.zip)

georgehines, 2016-04-13:
It seems from opening the log as text that the INFO statements regarding blackbox are being written to the beginning of the logfile

georgehines, 2016-04-13:
removing those INFO outputs results in a valid FMT header block, but a bad message_type is encountered. In FlightPlot, error is

`FormatErrorException: Unknown message type: 2`

In sdlog2_dump.py, error is

`Exception: Invalid header at 23070 (0x5A1E): 6C 47, must be A3 95`

In my direct matlab importer (ecl/matlab/analysis/importPX4log.m), inserting a breakpoint at first error indicates that a bad message type (191) comes in after the first AIRS message

georgehines, 2016-04-13:
manually deleting these text writes allows me to import the log successfully.

julianoes, 2016-04-14:
This line gets printed into the logfile somehow:
https://github.com/PX4/Firmware/blob/master/src/modules/sdlog2/sdlog2.c#L537

LorenzMeier, 2016-04-18:
Thanks, fixed on master.

closed 2016-04-18 by LorenzMeier

julianoes, 2016-04-19:
I see this commit b90f6dc15738b68a73fe011d6161599a4dbd5c52 but I don't understand it :)

bellahduan, 2017-10-12:
@LorenzMeier if it is Fixed wing,the .px4log download from QGC ,It can not use to analysis . but if it is Mulrotor wing .the .px4log download from QGC ,it is working well . is there any way to solve the problem ?

LorenzMeier, 2017-10-12:
The current logging format is ulog / .ulg - the old format is not supported any more. Which software version are you running?

LorenzMeier, 2017-10-12:
Recent fix here: https://github.com/PX4/Firmware/pull/8058