PX4/PX4-Autopilot #4270: Emergency Stop is not immediate

tops4u, 2016-04-14:
When issuing the Emergency Stop there is a delay. As can be seen in the attached File the Delay is mostly about 1.3 secs before the motor would spin down. IMHO an Emergency Stop should be immediate. In some cases it can be heard that the Motor spins down a bit just after flipping the switch but then stays on the reduced speed for some time before finally stopping. 
Setup:
The RC Connection is a Spektrum DSMx with 22ms (DX7s TX, AR 10000 RX), then there is a PPM Encoder Board as designed by DIYDRONESCOMMUNITY (John Arne and Olivier Adler) attached to a AUAV-X2. ESC are Hobbywing X-Rotor 40A running on 6S. 

[EmergencyStop.zip](https://github.com/PX4/Firmware/files/219937/EmergencyStop.zip)

kd0aij, 2016-04-14:
This should have fixed the issue: https://github.com/PX4/Firmware/pull/3753
It went into master on 23 Feb

tops4u, 2016-04-14:
Ok, I was not aware that 3753 also fixed the delay. I'll retest on the current Daily Master. Above was running on 1.2 Stable.

tops4u, 2016-04-20:
Checked with current Master - ok.

closed 2016-04-20 by tops4u