PX4/PX4-Autopilot #4275: Request: Servo Cam Gimbal Stabilization

tubeme, 2016-04-15:
Hi everybody,

I decided to set up a mini servo gimbal and I found I could not stabilize it with PX4 like I do in APM. 

Did I miss something or this functionality is not yet implemented in PX4? Or is it with some app that I could not find. I remember somebody writing in the groups that it is possible with an app you start in NuttX. I could not find the post though.

Thanks

AndreasAntener, 2016-04-15:
@tubeme see here: http://dev.px4.io/advanced-gimbal-control.html

If you start the gimbal app and setup the mixer channels it will command the actuators according to attitude (from -180 to 180 degree). So you want to change the scale in the mixer channels to match your servo travel.

It's really basic. For more complete gimbal support there is a bit to be done still, I'd suggest to open issues for specific features you'd like to see. The main use of this app at the moment is to control an actual gimbal controller (e.g. SimpleBGC) with telemetry.

closed 2016-04-15 by AndreasAntener

tubeme, 2016-04-15:
This does not include actually stabilization as far as I see. I need external stabilization like BCG etc. If I need to implement a simple servo gimbal with mini FPV camera I cant stabalize it with px4. Can I somehow use the plane roll and pitch (making double lines in the mixer) and connect the roll and pitch servo of the gimbal? in order to have the same stabilization the plane it having. Of course the SP are different but still I will have some sort of stabilization.

In APM there is a check box to the 3 axis so you can turn on and off each axis stabilization. It works for only for the servo gimbals.

I'm just trying to make an FPV platform with least amount of gear ie. wight to be applicable for small planes 1m wing span.

AndreasAntener, 2016-04-15:
You can stabilize it with PX4, just start the gimbal app, configure the mixer, then tilt/roll your copter. It will move the servos accordingly.

tubeme, 2016-04-15:
Ohhhh Perfect :) Thanks