PX4/PX4-Autopilot #4276: Proposal: Retractable landing gear param.

tubeme, 2016-04-15:
Hello,

I'm trying to set up the retractable landing gear and automate it.

The only option I have so far is to manually pass through to the mixer aux 1 to the retractable landing gear servo. And then I plan on using a custom MAV command set servo in order to automate the retracts in mission.

It is not a bad idea to have a parameter to map RC channel for the retracts. Also automate the retracts in the take off and landing process like it is with the flaps.

tops4u, 2016-04-16:
Duplicate to: #4007 ?

tubeme, 2016-04-17:
Well not exactly because I'm talking about airplane but yes it is from the same procedure ie. the landing.

Looking at the landing procedure stand point this: https://github.com/PX4/Firmware/issues/4201

the retractable landing gear for the plane should be enabled at a certain part of the landing process.

If we use Parachiute landing then again the moment we have the parachude open the landing gear should be retracted in order for the plane to land on them given they can sustain 3 m/s falling speed.

Probably the same procedure for parachute landing will be for MC. Then you need the rectracts to extend in order to land on them.

For the airplane take off it is the same. For MC take off probably similar.

The most important point in this issue is to be able to assign a RC Channel to a dedicated Retractable landing gear parameter, because we need the RC Aux1-3 for 3 axis camera. So using RC Aux1-3 for retracts directing them to a servo out with the mixer eats up 1 channel from the camera.

It should be no different than the current implementation of Camera Retract which is implemented in gimbal app.

LorenzMeier, 2016-05-22:
We will do this for 1.5.

LorenzMeier, 2016-10-23:
Implemented for multicopters. I'm closing this - anyone wanting to have automatic landing gear for fixed wing: Please enlist the help of a developer and send a PR.

closed 2016-10-23 by LorenzMeier