PX4/PX4-Autopilot #4280: Switching from off board mode to posHold does not set Z set point (SITL)

ecmnet, 2016-04-15:
In SITL, I switch from OFFBOARD (which controls via NED) to POSHOLD via MAVLink-Command. Doing so, I expected the current altitude as initial set point for NED-Z in POSHOLD. However the Z-setpoint is adjusted to the current altitude until the device has landed. XY seem to be ok - the xy setpoints do not change.

LorenzMeier, 2016-04-18:
@tumbili Something for you to look at.

tumbili, 2016-06-29:
@ecmnet Can you please post a log here?

ecmnet, 2016-06-29:
Unfortunately not, as the behavior has changed with the latest master updates. Currently RTL is triggered when switching from OFFBOARD to POSHOLD. I would not have expected that as well as, RC CHECK is set to Joystick/No RC check.

tumbili, 2016-06-30:
@ecmnet Hm, I'm not very familiar with the offboard logic. Does it give you any messages why it switched into RTL?

ecmnet, 2016-06-30:
No via MAVLink. Have to check logs.

ecmnet, 2016-06-30:
@tumbili Here one log: http://logs.uaventure.com/view/XqiCZqgRWyfBd2UVNy2NCF. All happens during phase 'AltHold'. Notice the altitude change via off board control in that phase. It is also in that phase, that I try to switch on PosHold. Instead of doing this, RTL is triggered, POSHold is activated after landing. Could it be that some parameters are wrong set? I tried that with posix_sitl_default jmavsim as well as with posix_sitl_lpe  jmavsim.

AndreasAntener, 2016-06-30:
I had a quick look. Main state does change to POSCTL, but nav state to RCRECOVER, which means it thinks it lost RC, although the log doesn't show the RC loss (but the failsafe).

During offboard it also went into failsafe for just an instance. There the nav state was set to LAND, which also means that it had the RC loss flag set.

AndreasAntener, 2016-06-30:
FYI, RC loss is not shown in the log because that flag is only set by actual RC input.

ecmnet, 2016-06-30:
Maybe the MAVLink RTL initiated message should contain the reason (e.g. instead the rising altitude).

AndreasAntener, 2016-07-04:
@ecmnet I did a test now with latest master, sending position setpoint in offboard an then switching to posctl. Once while hovering still and once while it was still in a climb to reach Z SP. Cannot reproduce the problem.

http://logs.uaventure.com/view/gkhyp6nLJoyWbJKY98dhVc
[image]

It holds the Z SP just fine.

The only thing I can think of right now that would cause your issue is that your joystick in center position is giving a signal out of the dead zone and therefore altitude hold is not active. If you can get your setup running again you might want to check what the manual input on Z is doing.

ecmnet, 2016-07-04:
@AndreasAntener Tested it again with latest master and a Taranis FrSky attached via USB and it works for me also now. Closing this.

closed 2016-07-04 by ecmnet