PX4/PX4-Autopilot #4289: Support for DJI guidance module for collision avoidance/ visual odometry

jgoppert, 2016-04-18:
I have already forked the official DJI guidance module:
https://github.com/jgoppert/Guidance-SDK-ROS

It is really just a c++ demo program wrapped into ROS. 

The DJI guidance is a 5 direction stereo vision system with ultrasonic sensors using an FPGA for processing. It then sends obstacle distance and visual odometry over usb to ROS. It can also be configued to send depth images.

I plan to rewrite the official DJI ros node so that it publishes topics cleanly and the startup is configurable in the launch file.
Once I get this done I just need to know how to modify mavros to send the data cleanly to the px4 firmware so that we can use the data.

This could be handled with the following mavlink messages:

### Messages
```
VISION_POSITION_ESTIMATE ( 102 )
VISION_SPEED_ESTIMATE ( 104 )
DISTANCE_SENSOR ( #132 )
```

But this means we will need to make sure mavlink_receiver can publish all of the distance sensor data from the guidance (5 directions).

jgoppert, 2016-04-18:
I think consideration needs to be taken to make sure the interface for this device is the exact same as the interface for the ETHZ solution.

jgoppert, 2016-04-18:
If the controller for collision avoidance is going to rely on the DISTANCE_SENSOR data, the behavior is going to be very basic, if we want to implement a VFH or potential field method, we will want a more exact direction of the closest point. The DJI guidance module is not setup to enable this, but the ETH device could.

TSC21, 2016-04-18:
@jgoppert count on me to aid you on this. `mavlink_receiver` is already ready to receive data from `DISTANCE_SENSOR`, we just need to define the proper ID numbers for each of the 5 sensors. Also, mavros is also prepared to load data into the Pixhawk through the `distance_sensor` plugin (I wrote it and tested it myself).
The only doubt I have now is how the the FCU will be handling each of the distance_sensor publishing. I assume it is using a multisubsriber/multipublisher framework.

TSC21, 2016-04-18:
> Once I get this done I just need to know how to modify mavros to send the data cleanly to the px4 firmware so that we can use the data.

This is already done on the `distance_sensor` plugin. I can guide on how to use it if you want.

TSC21, 2016-04-18:
One thing: we have a plugin for publishing distance using mavros, but right now `VISION_SPEED_ESTIMATE` is not yet supported on the firmware. We can readd it to LPE, as it was one of the things I've done and made a PR to when you had that initial branch of LPE.

khalfan7, 2016-07-17:
@TSC21 You got the DJI guidance to work on PixHawk? With collision avoidance?

LorenzMeier, 2016-12-26:
Will interface general VIO framework.

closed 2016-12-26 by LorenzMeier

AlexisTM, 2017-08-24:
Also, the DJI Guidance works badly without a DJI flight controller connected to it. 
 
If you want to use it with the Pixhawk, you need the send the IMU information of the Pixhawk to the Guidance system with no information about it.