PX4/PX4-Autopilot #4294: Gazebo 6.6 SITL doesn't find Eigen files Ubuntu

jwerner37, 2016-04-18:
@LorenzMeier @nicolaerosia

Same Eigen issue/error as https://github.com/PX4/Firmware/issues/4195  but on Ubuntu 14.04 this time and with Gazebo 6.6 - the new default Gazebo6 version.  Having trouble rolling back to 6.5 (not available anymore?) and FindEigen.cmake or FindEigen3.cmake not found on my system to update  CMAKE_MODULE_PATH.

Any tips?  besides never updating a week before project is due....  Thanks!

LorenzMeier, 2016-04-18:
Have you installed `libeigen3-dev`?

You can check out revision d7f998c3e28ecbce0dd86a0ab34271d78bcecf78 in Tools/sitl_gazebo manually.

LorenzMeier, 2016-04-18:
I've put a workaround on master. If it doesn't work, the last commits in sitl_gazebo should teach you how to work around it.

closed 2016-04-18 by LorenzMeier