PX4/PX4-Autopilot #4297: slow yaw in gazebo simulation

ChristophTobler, 2016-04-19:
I have just pulled master and I wanted to reproduce https://github.com/PX4/Firmware/issues/4191. Now however, yaw is very slow in the gazebo simulation (make posix gazebo_iris_opt_flow and make posix_sitl_default gazebo) but it works in jmavsim. I use gazebo verison 6.5.1.

ChristophTobler, 2016-04-19:
@LorenzMeier Seems to be this one https://github.com/PX4/sitl_gazebo/commit/51f69f61ad45b521ae2b57878345c71a13f485dd or what was the reason for that?
But even when I do: git checkout 58b61d68a618 (8 commits before) the simulation feels a bit weird. E.g. Altitude Control does not really work -> the iris goes up and down

ChristophTobler, 2016-06-01:
this commit slows down yaw: https://github.com/PX4/sitl_gazebo/commit/63cc44d84f4391133fdabb07e87da50aec5a5944

LorenzMeier, 2016-06-01:
Can you submit a PR to fix it? Or at least share the exact diff? Thanks!

ChristophTobler, 2016-06-01:
this is the exact diff. But I can also submit a PR.

LorenzMeier, 2016-12-28:
Stale / fixed.

closed 2016-12-28 by LorenzMeier