PX4/PX4-Autopilot #4304: New uorb topic generation issue on Snapdragon

tumbili, 2016-04-19:
The https://github.com/PX4/Firmware/pull/4197 branch does not work on Snapdragon.
After the sensors app is started on the DSP starting the commander or ekf2 app will make the DSP crash.
I have reduced the startup script of the DSP and I tried starting the apps manually via qshell. 
The branch which I am using is this one https://github.com/PX4/Firmware/tree/newLoggerSnapi
Here is what I get on the Posix side:
```
qshell sensors start
qshell commander start
```
Output:
```
qshell sensors start
pxh> qshell commander start
pxh> /home/cberg/Desktop/r00015/oe-core/build/tmp-eglibc/work/cortexa8hf-vfp-neon-linux-gnueabi/adsprpc/1.0-r0/adsprpc-1.0/src/listener_android.c:129:listener protocol failure -1
/home/cberg/Desktop/r00015/oe-core/build/tmp-eglibc/work/cortexa8hf-vfp-neon-linux-gnueabi/adsprpc/1.0-r0/adsprpc-1.0/src/listener_android.c:133::error: -1: 0 == (nErr = __QAIC_HEADER(adsp_listener_next_invoke)( ctx, nErr, 0, 0, &ctx, &handle, &sc, bufs->inbufs, inBufsAllocated, bufs->inbufLenReqs, MAX_BUFS, bufs->outbufLenReqs, MAX_BUFS))
/home/cberg/Desktop/r00015/oe-core/build/tmp-eglibc/work/cortexa8hf-vfp-neon-linux-gnueabi/adsprpc/1.0-r0/adsprpc-1.0/src/listener_android.c:192:listener thread exiting with code -1
/home/cberg/Desktop/r00015/oe-core/build/tmp-eglibc/work/cortexa8hf-vfp-neon-linux-gnueabi/adsprpc/1.0-r0/adsprpc-1.0/src/fastrpc_apps_user.c:224::error: -1: 0 == (nErr = remotectl_open(name, (int*)ph, dlerrstr, sizeof(dlerrstr), &dlerr))
/home/cberg/Desktop/r00015/oe-core/build/tmp-eglibc/work/cortexa8hf-vfp-neon-linux-gnueabi/adsprpc/1.0-r0/adsprpc-1.0/src/listener_android.c:194:listener thread failed to cleanly shutdown.  This is ok durring process exit.
/home/cberg/Desktop/r00015/oe-core/build/tmp-eglibc/work/cortexa8hf-vfp-neon-linux-gnueabi/adsprpc/1.0-r0/adsprpc-1.0/src/fastrpc_apps_user.c:224::error: -1: 0 == (nErr = remotectl_open(name, (int*)ph, dlerrstr, sizeof(dlerrstr), &dlerr))
/home/cberg/Desktop/r00015/oe-core/build/tmp-eglibc/work/cortexa8hf-vfp-neon-linux-gnueabi/adsprpc/1.0-r0/adsprpc-1.0/src/fastrpc_apps_user.c:224::error: -1: 0 == (nErr = remotectl_open(name, (int*)ph, dlerrstr, sizeof(dlerrstr), &dlerr))
/home/cberg/Desktop/r00015/oe-core/build/tmp-eglibc/work/cortexa8hf-vfp-neon-linux-gnueabi/adsprpc/1.0-r0/adsprpc-1.0/src/fastrpc_apps_user.c:224::error: -1: 0 == (nErr = remotectl_open(name, (int*)ph, dlerrstr, sizeof(dlerrstr), &dlerr))
/home/cberg/Desktop/r00015/oe-core/build/tmp-eglibc/work/cortexa8hf-vfp-neon-linux-gnueabi/adsprpc/1.0-r0/adsprpc-1.0/src/fastrpc_apps_user.c:224::error: -1: 0 == (nErr = remotectl_open(name, (int*)ph, dlerrstr, sizeof(dlerrstr), &dlerr))
ERROR ERROR: Getting Bulk data from fastRPC link
/home/cberg/Desktop/r00015/oe-core/build/tmp-eglibc/work/cortexa8hf-vfp-neon-linux-gnueabi/adsprpc/1.0-r0/adsprpc-1.0/src/fastrpc_apps_user.c:224::error: -1: 0 == (nErr = remotectl_open(name, (int*)ph, dlerrstr, sizeof(dlerrstr), &dlerr))
/home/cberg/Desktop/r00015/oe-core/build/tmp-eglibc/work/cortexa8hf-vfp-neon-linux-gnueabi/adsprpc/1.0-r0/adsprpc-1.0/src/fastrpc_apps_user.c:224::error: -1: 0 == (nErr = remotectl_open(name, (int*)ph, dlerrstr, sizeof(dlerrstr), &dlerr))
/home/cberg/Desktop/r00015/oe-core/build/tmp-eglibc/work/cortexa8hf-vfp-neon-linux-gnueabi/adsprpc/1.0-r0/adsprpc-1.0/src/fastrpc_apps_user.c:224::error: -1: 0 == (nErr = remotectl_open(name, (int*)ph, dlerrstr, sizeof(dlerrstr), &dlerr))
/home/cberg/Desktop/r00015/oe-core/build/tmp-eglibc/work/cortexa8hf-vfp-neon-linux-gnueabi/adsprpc/1.0-r0/adsprpc-1.0/src/fastrpc_apps_user.c:224::error: -1: 0 == (nErr = remotectl_open(name, (int*)ph, dlerrstr, sizeof(dlerrstr), &dlerr))

```

Here is what I get from the debugging terminal of the DSP:
```
[08500/00]  19:55.418  HAP:35:stack size: 1024 passed stacksize(2000)  0199  px4_qurt_tasks.cpp
[08500/00]  19:55.418  HAP:35:setting the thread[sensors] stack size to[8192]  0203  px4_qurt_tasks.cpp
[08500/02]  20:05.907  Process Sensor launched with ID=1   0130  main.c
[08500/02]  20:05.912  mmpm_register: MMPM client for USM ADSP core 12  0117  UltrasoundStreamMgr_Mmpm.cpp
[08500/02]  20:05.913  ADSP License DB: License validation function with id 164678 stored.  0280  adsp_license_db.cpp
[08500/02]  20:05.913  AvsCoreSvc: StartSvcHandler Enter  0518  AdspCoreSvc.cpp
[08500/02]  20:05.913  AdspCoreSvc: Started successfully  0534  AdspCoreSvc.cpp
[08500/02]  20:05.916  DSPS INIT  0191  sns_init_dsps.c
[08500/02]  20:05.916  INIT DONE  0224  sns_init_dsps.c
[08500/02]  20:05.916  Sensors Init : waiting(1)  0160  sns_init_dsps.c
[08500/02]  20:05.916  INIT DONE  0224  sns_init_dsps.c
[08500/02]  20:05.916  THRD CREATE: Thread=0x38 Name(Hex)= 53, 4e, 53, 5f, 53, 4d, 47, 52  0186  qurt_elite_thread.cpp
[08500/02]  20:05.916  THRD CREATE: Thread=0x37 Name(Hex)= 53, 4e, 53, 5f, 53, 41, 4d, 0  0186  qurt_elite_thread.cpp
[08500/02]  20:05.917  THRD CREATE: Thread=0x36 Name(Hex)= 53, 4e, 53, 5f, 53, 43, 4d, 0  0186  qurt_elite_thread.cpp
[08500/02]  20:05.917  THRD CREATE: Thread=0x35 Name(Hex)= 53, 4e, 53, 5f, 50, 4d, 0, 0  0186  qurt_elite_thread.cpp
[08500/02]  20:05.917  THRD CREATE: Thread=0x34 Name(Hex)= 53, 4e, 53, 5f, 53, 53, 4d, 0  0186  qurt_elite_thread.cpp
[08500/02]  20:05.917  THRD CREATE: Thread=0x33 Name(Hex)= 53, 4e, 53, 5f, 44, 45, 42, 55  0186  qurt_elite_thread.cpp
[08500/02]  20:05.917  Sensors Init : ///////////init once completed///////////  0169  sns_init_dsps.c
[08500/02]  20:05.917  Sensors DIAG F3 Trace Buffer Initialized  0260  sns_init_dsps.c
[08500/02]  20:05.917  loading BLSP configuration  0189  blsp_config.c
[08500/02]  20:05.917  INIT DONE  0224  sns_init_dsps.c
[00053/03]  20:05.918  Unsupported algorithm service id 0  0953  sns_scm_ext.c
[08500/02]  20:05.920  INIT DONE  0224  sns_init_dsps.c
[08500/02]  20:05.920  INIT DONE  0224  sns_init_dsps.c
[08500/02]  20:05.920  INIT DONE  0224  sns_init_dsps.c
[08500/02]  20:06.141  HAP:110:cannot find /oemconfig.so  0141  load.c
[08500/03]  20:06.141  HAP:111::error: -1: 0 == dynconfig_init(&conf, "security")   0696  sigverify.c
[08500/02]  20:06.142  HAP:74:cannot find /voiceproc_tx.so  0141  load.c
[08500/02]  20:06.148  HAP:74:cannot find /voiceproc_rx.so  0141  load.c
[08500/02]  20:06.150  HAP:159:unable to open the specified file path  0167  file.c
[08500/04]  20:06.150  failed to open /usr/share/data/adsp/blsp.config  0204  blsp_config.c
[08500/04]  20:06.150  QDSP6 Main.c: blsp_config_load() failed  0261  main.c
[08500/02]  20:06.150  Loaded default UART-BAM mapping  0035  blsp_config.c
[08500/02]  20:06.150  UART tty-1: BAM-9  0043  blsp_config.c
[08500/02]  20:06.150  UART tty-2: BAM-6  0043  blsp_config.c
[08500/02]  20:06.150  UART tty-3: BAM-8  0043  blsp_config.c
[08500/02]  20:06.150  UART tty-4: BAM-2  0043  blsp_config.c
[08500/02]  20:06.150  UART tty-5: BAM N/A  0048  blsp_config.c
[08500/02]  20:06.150  UART tty-6: BAM N/A  0048  blsp_config.c
[00053/03]  20:15.920  Cancel SNS_EM_TIMER_TYPE_ONESHOT  0306  sns_em.c
[00053/03]  20:15.920  SMGR not ready!  5529  sns_sam.c
[08500/02]  20:18.056  INIT DONE  0224  sns_init_dsps.c

```

tumbili, 2016-04-19:
@kd0aij @dagar FYI

mcharleb, 2016-05-02:
It looks like the DSP reset

julianoes, 2016-05-03:
Quoting what I said in https://github.com/PX4/Firmware/pull/4366#issuecomment-215406367:

> I narrowed the problem down but I can't figure out why exactly the crash happens.
> 
> The failing line is:
> https://github.com/PX4/Firmware/blob/ec4d2fa8de963e1d9379aa32e57b1723619b7b0d/src/modules/commander/commander.cpp#L1740
> 
> The pointer and value at ~~`&sensors.barometer_failure`~~ `&sensors.barometer_timestamp` are correct.
> 
> The crash is not happening when `hrt_elapsed_time()` is used without the `volatile` keyword:
> The declaration is here: https://github.com/PX4/Firmware/blob/ec4d2fa8de963e1d9379aa32e57b1723619b7b0d/src/drivers/drv_hrt.h#L103
> 
> Alternatively, you can work around the bug like below but it won't prevent the same thing from happening anywhere else:
> 
> ```
> const hrt_abstime temp_timestamp = sensors.baro_timestamp[0];
> if (hrt_elapsed_time(&temp_timestamp) < FAILSAFE_DEFAULT_TIMEOUT) {
> ```
> 
> @mcharleb, @LorenzMeier: any idea why Snapdragon would crash when calling the function which uses  has a `volatile` argument?
> Or asked differently: what is different about a variable from a uORB struct versus the `temp_timestamp` variable.

bkueng, 2016-05-03:
The plan is to avoid `#pragma pack` for the new logger, which will resolve these issues, so I don't think we should worry about this stuff.

julianoes, 2016-05-03:
So without `#pragma pack` we should not see this crash?

bkueng, 2016-05-03:
No I'm pretty sure we should not. The new logger doesn't really do any changes to the topic structs, except for adding a timestamp and the `#pragma pack`. Packing really goes against compiler fundamentals. They assume alignment, and if you go against that forcefully, you have to expect that bad things will happen.

LorenzMeier, 2016-05-03:
@dagar @bkueng What about a stable sort on fields and for nested messages adding padding bytes onto the next 32bit boundary?

bkueng, 2016-05-03:
@LorenzMeier AFAIK @dagar is planning to do that, not sure how far he progressed already

closed 2016-05-19 by LorenzMeier