PX4/PX4-Autopilot #4309: Estimators Clarification.

tubeme, 2016-04-20:
Hi guys,

I read the rc init.d scripts and have something I cannot understand:

From Plane def app rc: Firmware/ROMFS/px4fmu_common/init.d/rc.fw_apps

we have 

#!nsh
#
# Standard apps for fixed wing
#

#
# Start the attitude and position estimator
#
ekf_att_pos_estimator start

#
# Start attitude controller
#
fw_att_control start
fw_pos_control_l1 start

#
# Start Land Detector
#
land_detector start fixedwing

#
# Misc apps
#

Which means that the default estimator for FW is EKF.

From MC Firmware/ROMFS/px4fmu_common/init.d/rc.mc_apps we have:

#!nsh
#
# Standard apps for multirotors:
# att & pos estimator, att & pos control.
#

#---------------------------------------
# Estimator group selction
#
# INAV
if param compare SYS_MC_EST_GROUP 0
then
	attitude_estimator_q start
	position_estimator_inav start
fi

# LPE
if param compare SYS_MC_EST_GROUP 1
then
	attitude_estimator_q start
	local_position_estimator start
fi

# EKF
if param compare SYS_MC_EST_GROUP 2
then
	ekf2 start
fi
#---------------------------------------

if mc_att_control start
then
else
	# try the multiplatform version
	mc_att_control_m start
fi

if mc_pos_control start
then
else
	# try the multiplatform version
	mc_pos_control_m start
fi

#
# Start Land Detector
#
land_detector start multicopter

For the estimator here we have three options as EKF2 is the last one.

So now my question is what is the situation for  VTOL?
Firmware/ROMFS/px4fmu_common/init.d/rc.vtol_apps

#!nsh
#
# Standard apps for vtol:
# att & pos estimator, att & pos control.
#
#

#---------------------------------------
# Estimator group selction
#
# INAV
if param compare SYS_MC_EST_GROUP 0
then
	attitude_estimator_q start
	position_estimator_inav start
fi

# LPE
if param compare SYS_MC_EST_GROUP 1
then
	attitude_estimator_q start
	local_position_estimator start
fi

# EKF
if param compare SYS_MC_EST_GROUP 2
then
	ekf2 start
fi
#---------------------------------------

vtol_att_control start
mc_att_control start
mc_pos_control start
fw_att_control start
fw_pos_control_l1 start

#
# Start Land Detector
# Multicopter for now until we have something for VTOL
#
land_detector start vtol

I see estimators only for the MC mode of the VTOL. What is the estimator when the VTOL is in FW? I don't see ekf as option nor we start the ekf in the VTOL.

Can somebody clarify the situation?

Also I cannot understand something. Is EKF2 attitude and position estimator at the same time? This is how I understand it BTW...

devbharat, 2016-04-20:
Yes EKF2 estimates both attitude and position.

julianoes, 2016-04-22:
@tubeme: Can you please put \`\`\` around your pasted code? Otherwise, it's hard to read.

tubeme, 2016-04-22:
@julianoes Sorry for the mess... Couldn't format it. Now it is better.

julianoes, 2016-04-22:
See this page: http://dev.px4.io/advanced-switching_state_estimators.html

@tumbili: what about VTOL?

tubeme, 2016-04-22:
I saw this page. Actually it was added just after I posted this issue...

But again did not find the answer there about VTOL?

tumbili, 2016-05-03:
```
For multirotors and VTOL use the parameter SYS_MC_EST_GROUP to chose between the following configurations.
At the moment only the depriciated EKF estimator is used for planes (non VTOL). It will soon be replaced by EKF2. 
```

It is written like this in the dev guide.

closed 2016-05-03 by tumbili