PX4/PX4-Autopilot #4310: Eagle Tree Airspeed v3 PROBLEM.

tubeme, 2016-04-20:
Hello,

I'm just testing the Egle Tree airpseed sensor V3 which is i2c sensor.

It actually works out of the box!! Very nice. Just plug it in the i2c and PX4 loads the driver automaticaly. Whithout the need of extras.txt and ets_airspeed start command.

But probably there is some problem detecting the airspeed is present in the pre flight check.

I think so, becase I calibrated all the sensors including Level Horizon.

I calibrated the Airspeed as well and it shows perfect airspeed readings when I blow in the pitot tube.

All the sensors are in green, everything seems normal.

But I receive sensor warning in the QGC "Critical: Not ready to fly: Sensors not set up correctly" and I get the red blinking led and I cannot arm the airplane.

In the start up I see no obvious problems... Here it is:

sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
[i] microSD mounted: /fs/microsd
param: selected parameter default file /fs/mtd_params
[param] Loaded: /fs/mtd_params
rgbled on I2C bus 2 at 0x55 (bus: 100 KHz, max: 100 KHz)
 1:   SYS_USE_IO: match
 2104: + SYS_AUTOSTART: match
px4io: CRCs match
dataman: Power on restart, data manager file '/fs/microsd/dataman' size is 103090 bytes
MS5611_SPI on SPI bus 1 at 3 (20000 KHz)
bst: no devices found
adc init done
ver hwcmp match: PX4FMU_V2
HMC5883_I2C on I2C bus 1 at 0x1e (bus: 100 KHz, max: 400 KHz)
hmc5883: no device on bus 1
mpu6000: no device on this bus
MPU6000 on SPI bus 1 at 4 (1000 KHz)
L3GD20 on SPI bus 1 at 1 (11000 KHz)
LSM303D on SPI bus 1 at 2 (11000 KHz)
meas_airspeed: no MS4525 airspeed sensor connected
Airspeed on I2C bus 1 at 0x75 (bus: 100 KHz, max: 100 KHz)
nsh: sf10a: command not found
px4io default PWM output device
mavlink_if0: mode: 0, data rate: 1200 B/s on /dev/ttyS1 @ 57600B
mavlink_if0: offboard mission init: ERROR
 157600:   SYS_COMPANION: match
mavlink_if1: mode: 3, data rate: 1000 B/s on /dev/ttyS2 @ 57600B
ver hwcmp match: PX4FMU_V2
px4flow [156:100]
px4flow: scanning I2C buses for device..
mavlink_if2: mode: 5, data rate: 800000 B/s on /dev/ttyACM0 @ 57600B
WARN  log buffer size: 12288 bytes
[i] FIXED WING
[i] Mixer: /etc/mixers/AETR.main.mix on /dev/pwm_output0
pwm: reading disarmed value of zero, disabling disarmed PWM
[i] Mixer: /etc/mixers/pass.aux.mix on /dev/pwm_output1

NuttShell (NSH)
nsh> mavlink_if1: Disabling hardware flow control
INFO  filter ref off: baro_alt: -557.3599
commander_low_prio: settings saved.
WARN  Not ready to fly: Sensors not set up correctly

Any ideas?

tubeme, 2016-04-20:
I forgot to mention the only thing I had to do is to make the Eagle Tree sensor in metric units. I suppose if the sensor is in British units it will show different reading. Anyway it was important to mention i was consistent with the install.

tubeme, 2016-04-21:
Still cannot find the source of the problem...

LorenzMeier, 2016-04-21:
Are you sure its related to the airspeed sensor? How high is your battery voltage when you are testing?

tubeme, 2016-04-22:
I'm sure it is not the battery. It is 3S bat I'm using and it is in range, there is no low voltage alarm. In QGC the meter shows 50+% for the battery. Strangely all the sensors are working properly including the airspeed. I get GPS lock. The mag is working the internal and the external as well. All the sensors ere within margins because I get 4 green lights in the Sensors calibration tab after calibration... Durign calibration there were no signs that some of the sensors are misbehaving...The Eagle Tree Airspeed is the only difference in this install from the default setup. I'm trying to find something in the logs but nothing except this message: Not ready to fly: Sensors not set up correctly.

I will try the original airspeed later today to see if this is the reason. .

tubeme, 2016-04-22:
It is not the airspeed...with the original one it is the same thing. It is something else that I cannot detect...

tubeme, 2016-04-23:
I'm lost... I reset a couple of times the frame. The same error: Sensors not setup correctly.

What could be not set up correctly? I did this setup a hundreds of times the same way...

I do not use custom mixer nor custom apps. Just the default AETR frame.

The error Not ready to fly: Sensors not set up correctly. is obvious even prior sensors calibration. After sensors calibration it did not disappear so it is not calibration related. All the sensors work perfect... 

I changed the power cable from the brick in order to confirm there is no problem with the volt and amp readings from the brick. The same story.

Any help will be appreciated.

tubeme, 2016-04-23:
Is this OK?

INFO filter ref off: baro_alt: -557.3599

baro altitude -557 meters?

LorenzMeier, 2016-04-25:
Closing this as it seems related to something else.

closed 2016-04-25 by LorenzMeier