PX4/PX4-Autopilot #4313: Eigen Include Error

tasarmiento, 2016-04-20:
In file included from /Users/TAS/PX4_src/sitl_gazebo/include/gazebo_multirotor_base_plugin.h:29:
/Users/TAS/PX4_src/sitl_gazebo/include/common.h:22:10: fatal error: 
      'Eigen/Dense' file not found
#include <Eigen/Dense>
         ^
1 error generated.

In file included from /Users/TAS/PX4_src/sitl_gazebo/src/gazebo_controller_interface.cpp:22:
/Users/TAS/PX4_src/sitl_gazebo/include/gazebo_controller_interface.h:23:10: fatal error: 
      'Eigen/Eigen' file not found
#include <Eigen/Eigen>
         ^
1 error generated.

Can be solved by updating the path to:  #include <eigen3/Eigen/Dense>.
Believe this was solved back in January, but the issue resurfaced today.

LorenzMeier, 2016-04-20:
What is the exact diff in your local folder? Sorry about the regression - some well intended cleanup didn't apparently really work.

tasarmiento, 2016-04-20:
Lorenz,

The only thing missing was eigen3 before Eigen in the path.  I believe this
is the same issue that happened back in Jan (posted here:
https://groups.google.com/forum/#!searchin/px4users/eigen/px4users/3ttWk0GpwtU/9KTqh7-GFgAJ
), but I just forked that repository today to start working and had the
same issue with make.

After fixing the eigen issue I ran into the same optic flow sensor
follow-on issue as #4195, but removing Gazebo7 and installing 6 fixed the
issue.  Somewhere in your previous instructions they noted Mac users must
install Gazebo7, but I believe you've since removed that comment.

Thanks for all the great work on this SITL simulation.

-Traci.

LorenzMeier, 2016-04-21:
Can you please give me the output of git diff in Tools/sitl_gazebo? I don't understand what you changed.

tasarmiento, 2016-04-21:
Sorry it was unclear.  I did not make changes in Firmware/Tools/sitl_gazebo

Here is the output of git diff from ~/PX4_src/sitl_gazebo where I made the
changes:

*diff --git a/include/common.h b/include/common.h*

*index 5d97a44..9ad9984 100644*

*--- a/include/common.h*

*+++ b/include/common.h*

@@ -19,7 +19,7 @@

  */

-#include <Eigen/Dense>

+#include <eigen3/eigen/Dense>

 #include <gazebo/gazebo.hh>

 namespace gazebo {

*diff --git a/include/gazebo_controller_interface.h
b/include/gazebo_controller_interface.h*

*index 8ae5c69..f29716b 100644*

*--- a/include/gazebo_controller_interface.h*

*+++ b/include/gazebo_controller_interface.h*

@@ -20,7 +20,7 @@

 #include <boost/bind.hpp>

-#include <Eigen/Eigen>

-Traci.

LorenzMeier, 2016-04-25:
Ok, pushed something equivalent to master.

closed 2016-04-25 by LorenzMeier