PX4/PX4-Autopilot #4315: Update Racer Quad gains

LorenzMeier, 2016-04-21:
@kd0aij @pkocmoud Could you work out a new default set for the generic 250 config and help us getting it in the v1.3.0 release?

I honestly don't understand where the blocker is for getting them validated and out to the community.

kd0aij, 2016-04-21:
The problem I have is that my 250 gains seem to be far too high for the few people that have tried them. So it seems they won't make good defaults.

kd0aij, 2016-04-21:
@pkocmoud Can you try the gains posted here: https://gitter.im/kd0aij/dual_FC_S250AQ?at=571528bb548df1be102dfda3 on your S250? Maybe the Spedix can handle higher gains thant the QAV250

LorenzMeier, 2016-04-22:
I don't believe we really understand this yet. I'm pushing this out to next release.

closed 2016-05-22 by LorenzMeier