PX4/PX4-Autopilot #4316: Passthrough mode for Snapdragon on PixFalcon fails to output PWM

ndepal, 2016-04-21:
I am using a PixFalcon in passthrough mode to control the ESCs with the Snapdragon flight.
The PixFalcon refuses to talk to the ESCs, which beep every few seconds as a result.

I am seeing this on the current master (af02b860b3cc495f67a879088dca9f8b1edd58f9) with the following procedure:
Flash the firmware to the PixFalcon
```make px4fmu-v2_default upload```
With the SD card out, connect to the NuttShell. Set a normal airframe and reboot:
```
param set SYS_AUTOSTART 10020
reboot
```
Once rebooted, re-connect, press the safety switch and test the PWM:
```
pwm test -c 1 -p 1200
```
The motor turns, the ESCs are quiet.
Set the passthrough airframe and reboot (via the nsh command, __not__ power-cycling):
```
param set SYS_AUTOSTART 20000
reboot
```
Re-connect and test the PWM (safety switch not required)
```
pwm test -c 1 -p 1200
```
The motor turns, the ESCs are quiet. If I `reboot`, the PixFalcon's LED will now light up purple instead of blinking blue, but `pwm test` still works.

As soon as I power-cycle, `pwm test` no longer works (and neither does `snapdragon_rc_pwm`).
After stopping `pwm test` with `Ctrl-C` I get the following message:
```
pwm: PWM_SERVO_SET(0): Invalid argument
```
I see the same output printed in [`snapdragon_rc_pwm`](https://github.com/PX4/Firmware/blob/2f48317a9ea7b05d8c61b3d1c44f8229491350b1/src/drivers/snapdragon_rc_pwm/snapdragon_rc_pwm.cpp#L209), where `ret == -1` and `errno == 22`.

LorenzMeier, 2016-04-21:
@julianoes Can you have a quick look?

julianoes, 2016-04-21:
@ndepal: Did it work before on that Pixfalcon?

ndepal, 2016-04-21:
Yes it did. It's very strange behavior. It worked Tuesday night, but didn't on Wednesday morning, even though I didn't do anything besides unplugging the device over night.

But it does seem to be caused by that hard reboot.

Are there any parameters that are set by airframe 10020 that might survive a `reboot` but not a power-cycle?

julianoes, 2016-04-21:
There would be some parameters that get set if you selected the airframe with QGC which will also set `SYS_AUTOCONFIG`. However, I don't know how that would influence the startup with 20000.

ndepal, 2016-04-21:
If I set the airframe with QGC I can't get `pwm test` to work even once. But this is probably because I have to power-cycle the PixFalcon (and remove the SD card) to get back into the NuttShell. Additionally the LED then flashes red, presumably because the IMU calibration is lost?

julianoes, 2016-04-21:
@ndepal Did you do `param save` because I don't see it above?

julianoes, 2016-04-21:
I flashed current master on a Pixfalcon and tried both `SYS_AUTOSTART`:

**10020**
```
nsh> tone_alarm stop
nsh> param show SYS_AUTOSTART
Symbols: x = used, + = saved, * = unsaved
x + SYS_AUTOSTART [380,680] : 10020

 790 parameters total, 402 used.
nsh> pwm info
device: /dev/pwm_output0
channel 1: 900 us (alternative rate: 400 Hz failsafe: 900, disarmed: 900 us, min: 1100 us, max: 1950 us)
channel 2: 900 us (alternative rate: 400 Hz failsafe: 900, disarmed: 900 us, min: 1100 us, max: 1950 us)
channel 3: 900 us (alternative rate: 400 Hz failsafe: 900, disarmed: 900 us, min: 1100 us, max: 1950 us)
channel 4: 900 us (alternative rate: 400 Hz failsafe: 900, disarmed: 900 us, min: 1100 us, max: 1950 us)
channel 5: 0 us (default rate: 50 Hz failsafe: 1500, disarmed: 0 us, min: 1000 us, max: 2000 us)
channel 6: 0 us (default rate: 50 Hz failsafe: 1500, disarmed: 0 us, min: 1000 us, max: 2000 us)
channel 7: 0 us (default rate: 50 Hz failsafe: 0, disarmed: 0 us, min: 1000 us, max: 2000 us)
channel 8: 0 us (default rate: 50 Hz failsafe: 0, disarmed: 0 us, min: 1000 us, max: 2000 us)
channel group 0: channels 1 2
channel group 1: channels 5 6 7 8
channel group 2: channels 3 4
nsh> pwm test -c 1 -p 1200
pwm: Press CTRL-C or 'c' to abort.
pwm: User abort

nsh> 
```

**20000**
```
nsh> tone_alarm stop
nsh> param show SYS_AUTOSTART
Symbols: x = used, + = saved, * = unsaved
x + SYS_AUTOSTART [273,680] : 20000

 790 parameters total, 292 used.
nsh> pwm info
device: /dev/pwm_output0
channel 1: 900 us (alternative rate: 400 Hz failsafe: 900, disarmed: 900 us, min: 1100 us, max: 1950 us)
channel 2: 900 us (alternative rate: 400 Hz failsafe: 900, disarmed: 900 us, min: 1100 us, max: 1950 us)
channel 3: 900 us (alternative rate: 400 Hz failsafe: 900, disarmed: 900 us, min: 1100 us, max: 1950 us)
channel 4: 900 us (alternative rate: 400 Hz failsafe: 900, disarmed: 900 us, min: 1100 us, max: 1950 us)
channel 5: 1500 us (alternative rate: 400 Hz failsafe: 1500, disarmed: 0 us, min: 1000 us, max: 2000 us)
channel 6: 1500 us (alternative rate: 400 Hz failsafe: 1500, disarmed: 0 us, min: 1000 us, max: 2000 us)
channel 7: 0 us (alternative rate: 400 Hz failsafe: 0, disarmed: 0 us, min: 1000 us, max: 2000 us)
channel 8: 0 us (alternative rate: 400 Hz failsafe: 0, disarmed: 0 us, min: 1000 us, max: 2000 us)
channel group 0: channels 1 2
channel group 1: channels 5 6 7 8
channel group 2: channels 3 4
nsh> pwm test -c 1 -p 1200
pwm: Press CTRL-C or 'c' to abort.
pwm: User abort

nsh> 
```

ndepal, 2016-04-21:
did you power-cycle in between?

julianoes, 2016-04-21:
Aha, now I get the error as well.

ndepal, 2016-04-21:
With the help of Lorenz we found the issue. The ESCs need a failsafe signal before accepting anything else.
This PR fixes it for me: https://github.com/PX4/Firmware/pull/4317

julianoes, 2016-04-21:
Nice that you resolved it, however, does this fix the `PWM_SERVO_SET(x)` problem?

ndepal, 2016-04-21:
Yes. `PWM_SERVO_SET(x)` was shown because the ESCs didn't accept anything because they hadn't seen the failsafe signal, as far as I understand.

julianoes, 2016-04-21:
Ok, so the underlying logic there is wrong, need to dig there.

LorenzMeier, 2016-04-21:
We fixed it, PR incoming

closed 2016-04-21 by LorenzMeier

julianoes, 2016-04-21:
I'm fixing it as well :)