PX4/PX4-Autopilot #4322: Mauch Power module scaling

edsellar, 2016-04-21:
The Mauch power modules that are using a hall effect sensor are much more accurate from what is currently being used with each specific sensor coming with an input voltage divider as well as measured amps per volt to give an extremely accurate view of the exact sensor you have and its reported voltage and current instead of a general view for the sensor type. Where can we add this functionality? - it seems pretty straightforward on the APM stack in Mission Planner but I think I am having a disconnect with the scaling function here or where to enter these numbers.

[http://ardupilot.org/plane/docs/common-mauch-power-modules.html](url)

mhkabir, 2016-04-22:
Indeed. I have one of these as well, but I wasn't able to properly configure the current reading. Voltage was easier to get right with some empirical tuning.

@LorenzMeier Every one of these modules comes with a factory calibration result, which is emailed to the customer. It would be great to figure out how we can use this for accurate measurements.

[image]

AndreasAntener, 2016-04-22:
I believe you can calculate the PX4 values like this:

V_SCALING: "V divider" / (4096/3.3)
C_SCALING: "A/V" / (4096/3.3)

Could you give that a try?

edsellar, 2016-04-22:
Confirmed that V_SCALING works using the above. Will need to do more testing on C_SCALING but am confident should be ok if logic is the same. Not user friendly to input as needing to calculate prior to entry, etc but the functionality is there which is good. Thanks @AndreasAntener

mhkabir, 2016-04-23:
Voltage looks good. I had the current integrators going wild, but that might have something to do with a back-powering issue I just discovered on the airframe.

sanderux, 2016-05-03:
@mhkabir did your current integrator work by this setting eventually?

LorenzMeier, 2016-05-22:
Fixed in https://github.com/PX4/Firmware/pull/4603.

closed 2016-05-22 by LorenzMeier

khikmatovfarkhod, 2024-05-11:
Hello this did not fix anything. i am still getting null values @LorenzMeier @AndreasAntener