PX4/PX4-Autopilot #4329: OFFBOARD using LPE

MegaAdragon, 2016-04-22:
Hi,

I have a question regarding the OFFBOARD mode.

I want to use the OFFBOARD mode and send waypoints to my drone. I'm flying indoor using only optical flow and the Local Position Estimator.
To stream the MAVLink messages I use an iPad connected over BLE to the drone.

I used the C-UART-Interface (https://github.com/mavlink/c_uart_interface_example) as template and tried to implement the same thing on my iPad.
So basically I’m streaming a constant X,Y,Z SET_POSITION_TARGET_LOCAL_NED message. However, when I switch into OFFBOARD mode via RC (or via iPad), the drone starts accelerating really fast.
I tested it without props and while holding the drone with my hand:
http://logs.uaventure.com/view/SPXqCQfahAWj2nc9mEP5uN

I also did another test, where I switched into position hold before switching into OFFBOARD. The result seems to be the same:
http://logs.uaventure.com/view/E79cnzahir3pGqWFsZPeFJ

The plot for Local Position X,Y and Z clearly shows that the setpoint is still changing. The plot for the actuators shows how the drone is accelerating uncontrollably.
I expected that after sending a constant position target all the setpoints should show the constant position and the drone should hold its position there.

My goal is to use the iPad to let the drone hold its position at different waypoints (for example flying a square figure).

Can you give me some more information about the OFFBOARD mode? Especially how the LPE position is used.

jgoppert, 2016-04-25:
@MegaAdragon Looking into this, maybe related to #4352

vinhk, 2016-05-14:
Hi @MegaAdragon how did you implement Pixhawk with the LPE?

jgoppert, 2016-05-15:
@vinhk Set SYS_MC_EST_GROUP = 1 and reboot.

jgoppert, 2016-05-15:
I flew an auto mission with LPE recently so I believe this has been resolved @MegaAdragon? If there are problems with your offboard config it most likely isn't LPE related.

closed 2016-05-15 by jgoppert

vinhk, 2016-05-15:
James if you have time. Do you know how I can find out if my px4flow is
working with pixhawk? Is there a way to see data being sent to pixhawk from
optical flow?

jgoppert, 2016-05-15:
@vinhk  Yes, just plug in usb and start qgroundcontrol. Under the setup page you should see a px4flow display. Use this to focus the camera using text at the distance you intend to flight at.

vinhk, 2016-05-15:
I see. But without outputs from pixhawk, I'm not sure if it is working my
my pixhawk. I have successfully been able to view the display in
qgroundcontrol. Based on my knowledge of making px4flow work with pixhawk
is upload firmware and just connect serial wires from px4flow to pixhawk.

kd0aij, 2016-05-15:
With px4flow connected to pixhawk, QGC should
show MAVLINK_MSG_ID_OPTICAL_FLOW_RAD messages in the mavlink inspector

vinhk, 2016-05-15:
Ok thank you. I'll look into it.

LorenzMeier, 2016-05-16:
You need to interface it via I2C, *NOT* serial.