PX4/PX4-Autopilot #4330: Snapdragon Flight very high IMU noise and bad attitude estimation

ndepal, 2016-04-22:
I'm having the problem that the Snapdragon Flight's attitude estimation is performing very badly.
The snapdragon is mounted on a Bebop frame in a way that it takes advantage of the dampers that the Bebop has (although I'm not convinced they actually dampen the frequencies we need them to).

In flight, IMU measurements are extremely noisy, up to +/-100m/s^2, see this log: http://logs.uaventure.com/view/YbxWqciEHWJPFP8U4BRZYA

This was with the higher sampling of https://github.com/PX4/DriverFramework/pull/57.
The quad's performance in air is unsurprisingly quite terrible, the attitude seems to drift off significantly.

I have a few questions:
1) With the high sampling rate of 4kHz, are the measurements being down-sampled?
2) If they are, are they being filtered or simply windowed integration?
3) What other possibilities do we have to improve the measurements besides complicated mechanical damping? It seems this shouldn't be necessary, other platforms work even completely hard-mounted.

A likely related issue is that the attitude estimation is very poor even when the quad is stationary and mag fusion is off.

While my quad is stationary, I have an estimated yaw speed of about 20 deg/sec. I have uploaded a log here: http://logs.uaventure.com/view/UCmmvsKeorVQHENfYBWyZQ
I have recently calibrated the IMUs, but the mag calibration fails with `calibration failed: reading sensor` (I can see the mag measurements in QGC without a problem, though). Because of this failed mag calibration I disabled mag fusion in the EKF2 with `EKF_MAG_TYPE = None`.

With `EKF_MAG_TYPE = Magnetic heading`, the situation is much better. But it appears to simply be able to cover up a significant problem with the attitude estimation.

julianoes, 2016-04-22:
Thanks for reporting. I just did some thinking about the integration happening and I discovered a major flaw that I need to fix:

8 consecutive measurements are more or less published with the same timestamp. This means only every 8th dt is >> 0 and therefore we probably only really see every 8th sample.

I'll fix this next up.

priseborough, 2016-04-22:
Julien, I have a couple of questions about the integration method:

1) Are you using trapezoidal or rectangular integration?
2) What delta time do you assign to each trapezoid or rectangle?

julianoes, 2016-04-23:
@priseborough I'm working on the integration thing right now.

1) I think it is trapezoidal: `(val + _last_val) * dt * 0.5f`
2) The correct one :), at least with the changes I'm pushing soon.

LorenzMeier, 2016-04-25:
@ndepal Is this fixed?

ndepal, 2016-04-25:
I'm still seeing a very high IMU noise band:
http://logs.uaventure.com/view/sBGdZuKZoyPHDCZPg4Wr8K

What IMU values are being logged exactly? The ones integrated/downsampled to 250Hz? Or simply a subsampling of the raw values (i.e. taking every Xth raw value)?

LorenzMeier, 2016-04-25:
Your link doesn't work for me - is this the right link?

ndepal, 2016-04-25:
Yes, that should be the right link. Although I'm not seeing anything either.

Copy-pasted from the email:

Plots: http://logs.uaventure.com/view/sBGdZuKZoyPHDCZPg4Wr8K
Log: http://logs.uaventure.com/logmuncher/charts/sBGdZuKZoyPHDCZPg4Wr8K/logmuncher.log

julianoes, 2016-04-25:
@ndepal Yes, what you see in the log is not what ekf2 uses. You see 1/32 samples while the ekf2 see's the sum of 32 samples weighted by the dt (which should be pretty constant).

ndepal, 2016-04-25:
With https://github.com/PX4/Firmware/pull/4340 the in-flight performance issue is solved, attitude control is quite good now (confirmed by @tumbili). The attitude drift without mag fusion (second log in original post) is still the same.

LorenzMeier, 2016-04-26:
@priseborough @tumbili Could you look into the attitude drift when mag fusion is disabled? We would expect only a slow yaw drift, not a fundamental issue. This is very relevant for urban environments and it would be very helpful to resolve.

LorenzMeier, 2016-04-26:
I've replaced this issue with https://github.com/PX4/ecl/issues/108

closed 2016-04-26 by LorenzMeier

stu85010, 2016-08-02:
I have the same issue in v1.4.1rc2. The noise of accelerometer is very high.
Plots: http://logs.uaventure.com/view/qBkzZ9VQunNuRRsB6tGNYU#IMU_Acc_PLOT
Log: http://logs.uaventure.com/logmuncher/charts/qBkzZ9VQunNuRRsB6tGNYU/logmuncher.log

I'm also trying to mount Pixracer that only enabling the MPU9250 beside the snapdragon flight. But the waveform in sdlog is very smooth.