PX4/PX4-Autopilot #4338: Fix Jenkins CI reporting stdout issues

AndreasAntener, 2016-04-23:
- [ ] during rostest standard out gets filled with up to 100mb non printable chars
- [ ] there seems to be an issue with rostest reporting, non unicode chars in xml files

sanderux, 2016-05-13:
Jenkins now fails with following errors:
Failed to read test report file /build/jenkins/workspace/PX4 Firmware POSIX SITL PRs/test_results/rosunit-mavros_offboard_attctl_test.xml
org.dom4j.DocumentException: Error on line 2 of document file:///build/jenkins/workspace/PX4%20Firmware%20POSIX%20SITL%20PRs/test_results/rosunit-mavros_offboard_attctl_test.xml : Premature end of file. Nested exception: Premature end of file.

AndreasAntener, 2016-05-13:
That's because the tests failed. But it fails at random instances lately (one of the tests runs into a timeout), something must have changed recently.

AndreasAntener, 2016-12-03:
Seems this is no problem

closed 2016-12-03 by AndreasAntener