PX4/PX4-Autopilot #4341: Preflight check fail because of battery

tumbili, 2016-04-23:
http://discuss.px4.io/t/flying-wing-x5-setup-questions/349

See number 4.

tumbili, 2016-04-23:
@LorenzMeier What do you think of this? Could this be real?

jcooper21, 2016-04-23:
I can screen record it for you if you like.  I assure you its real, but could it be a parameter setting related to the power module?

tumbili, 2016-04-25:
@jcooper21 I just flashed master on my quad which has a Pixhawk and a 3DR battery module installed. I was not able to reproduce your error.

tumbili, 2016-04-25:
@jcooper21 Did you power your servo rail? The error you are seeing appear if either your brick or your servo rails is not powered or if you are still connected over USB.

tumbili, 2016-04-25:
Sorry, not true

tumbili, 2016-04-25:
@jcooper21 This error appears if your are powering the servo rails but not the brick. Do you have another power module to this this with?

LorenzMeier, 2016-04-25:
Moving this off the release as its a setup issue.

jcooper21, 2016-04-25:
Thanks I never would have guessed that could be the issue especially since this power module is brand new.  I have got an old power module to test with, so I will try that shortly. Thanks.

jcooper21, 2016-04-25:
I can confirm it was a bad power module.

closed 2016-04-25 by LorenzMeier

AlexWUrobot, 2023-02-03:
" [health_and_arming_checks] Preflight Fail: Battery unhealthy"
The same issue happened when running on the simulation.
I don't know how to solve this issue on the simulation.
Thanks a lot 
ref:https://discuss.px4.io/t/preflight-fail-battery-unhealthy/30678

AlexWUrobot, 2023-02-07:
It is very helpful to clean the `~/.ros`
ref: [Posix Sitl vehicles don't work correctly · Issue #18299 · PX4/PX4-Autopilot · GitHub](https://github.com/PX4/PX4-Autopilot/issues/18299)
ref: https://discuss.px4.io/t/simulation-take-off-motion-not-vertical/30705/4