PX4/PX4-Autopilot #4352: AltHold with LPE broken with latest master

ecmnet, 2016-04-25:
AltHold using LPE with latest master seems not to work properly anymore:

http://logs.uaventure.com/view/ec9QTt8uVfnTMXiatRxdNX#Local_Z_PLOT

LorenzMeier, 2016-04-25:
I'm not aware of any relevant changes. @jgoppert Can you have a look?

LorenzMeier, 2016-04-25:
We added an acceleration controller interface - are you using off board control?

ecmnet, 2016-04-25:
That is why I was also very curious - I did not expect any issues here.

jgoppert, 2016-04-25:
@ecmnet what did you havev MPC_ALT_MODE set to. If it is set to 0 you will see local z holding the set point. If you set it to MPC_ALT_MODE = 1 you will see dist bottom being regulated to the set point. I will fly master today to investigate further.

ecmnet, 2016-04-25:
@jgoppert MPC_ALT_MODE was set to 1. Flight was indoors above flat ground with LIDAR Sf10a. I also double-checked Raw LIDAR measurement which is working perfectly, but I some had sonar-timeouts.

jgoppert, 2016-04-25:
@ecmnet So you had good flight on master before with this param setup correct?  So I should look for recent changes in master?

jgoppert, 2016-04-25:
[image]

I analyzed your log. It seems very odd to me that the altitude set point is exactly the current altitude except for a couple of times where it is constant. This suggests it must be a recent change in mc_pos_control.  @LorenzMeier any ideas?

ecmnet, 2016-04-25:
@jgoppert Did my last test-flights April, 23 and had no issues up to then for many times.

jgoppert, 2016-04-25:
@ecmnet @LorenzMeier  It really points to this commit then: https://github.com/PX4/Firmware/commit/95faba391a8b6d891409679bc6dd94598b47f06b

This modified the altitude setpoint logic. We just have to figure out what broke.

LorenzMeier, 2016-04-25:
@ecmnet Are you using off board control or not? If you do, you need to make sure your flags are set up correctly and do not enable acceleration control: ```control_mode.flag_control_acceleration_enabled = !offboard_control_mode.ignore_acceleration_force && 
		  !status.in_transition_mode; ```

ecmnet, 2016-04-25:
@LorenzMeier @jgoppert I did not use off board control,  but I will double-check again and let you know.

jgoppert, 2016-04-25:
Is this only present with lpe, or other estimators as well?

ecmnet, 2016-04-25:
@jgoppert I did only check LPE as I am not using other estimators. Also I confirm, that I have not been using off board mode.

ecmnet, 2016-04-25:
@jgoppert Seems to be LPE. Although raw distance is correct, the LPE does not consider LIDAR measurements due to LIDAR beta faults. I cannot remember having those before.

ecmnet, 2016-04-25:
Update: After a beta fault, LIDAR does not recover. Estimation seems to be based on Baro only. Here's a long-time recording (LIDAR vs. LPE-Z):

[image]

jgoppert, 2016-04-25:
@ecmnet Long term drift in local z is normal, please plot alt.dist_bottom to see if that is drifting.

The reason local z should drift is that it follows baro, since this is the best measuremnt of global altitude, but the difference between the terrain and the global altitude is measured by the lidar, and that should be constant if the vehicle is stationary and the lidar is reading the same.

ecmnet, 2016-04-25:
@jgoppert Thanks for clarifying this. The difference between global altitude and terrain is constant:

[image]

jgoppert, 2016-04-25:
@ecmnet Ok, that looks good, but you still have issues while flying correct? So that must be related to the setpoint issue.

ecmnet, 2016-04-25:
@jgoppert I just tried to reset all params, re-calibrated sensors. Could it be that there was a parameter mess, when flashing the latest master? I'll do another test-flight tomorrow.

ecmnet, 2016-04-26:
After resetting parameters, altHold works again. Closing this issue

closed 2016-04-26 by ecmnet