PX4/PX4-Autopilot #4355: UBX Checksum errors

LorenzMeier, 2016-04-25:
@bkueng I get now a lot more UBX checksum error messages immediately after system boot on the serial console:

```
nsh> WARN  ubx checksum err
WARN  ubx checksum err
WARN  ubx checksum err
WARN  ubx checksum err
WARN  ubx checksum err
WARN  ubx checksum err
```

Do you have an idea where this is originating?

LorenzMeier, 2016-04-25:
I have a LEA 6H receiver, so a 2.5 years old model.

LorenzMeier, 2016-04-25:
```
nsh> gps status
WARN  protocol: UBX
WARN  port: /dev/ttyS3, baudrate: 38400, status: OK
WARN  sat info: disabled, noise: 81, jamming detected: NO
WARN  position lock: 0, satellites: 0, last update:  60.8780ms ago
WARN  lat: 449576611, lon: -787359380, alt: 10667875
WARN  vel: 0.00m/s, 0.00m/s, 0.00m/s
WARN  hdop: 99.99, vdop: 99.99
WARN  eph: 4294967.50m, epv: 4294967.50m
WARN  rate position:              5.00 Hz
WARN  rate velocity:              5.00 Hz
WARN  rate publication:           5.00 Hz
WARN  rate RTCM injection:        0.00 Hz
```

bkueng, 2016-04-25:
Is it reproducible? by 'a lot more' you mean you had them before as well? I did not add any new messages that are unsupported in older ubx protocol versions, only when in RTCM mode. On the Firmware side, I extended the ubx driver to configure the USB port and enable RTCM input.

Is it gone after startup? On startup it can be that it has the wrong baudrate, and the wrong output mode (NMEA by default). The messages then cannot be decoded correctly and you get checksum errors.

LorenzMeier, 2016-05-01:
Not seeing this any more.

closed 2016-05-01 by LorenzMeier

mhkabir, 2016-05-01:
I can observe this on Snapdragon Flight, in mini-dm. It seems intermittent though. M8N receiver.