PX4/PX4-Autopilot #4356: Snapdragon: some parameters don't want to stick

julianoes, 2016-04-25:
The param `PWM_DISARMED` just doesn't seem to stick/get saved on Snapdragon.

Also reported by @blandry.

julianoes, 2016-04-25:
The param seems to appear and work after `param set PWM_DISARMED 900` has been called once.

@LorenzMeier any idea why this is necessary?

LorenzMeier, 2016-04-25:
So you're saying that accessing it via MAVLink doesn't work, but on the shell does? Which shell? qshell or Linux side?

julianoes, 2016-04-25:
With `param set PWM_DISARMED 900` I mean the Linux side.

When QGC/mavlink is used, it is also `set` on the Linux side, however, the `save` happens on the QURT side. During the save, the written param is not in the list here:
https://github.com/PX4/Firmware/blob/master/src/modules/systemlib/param/param_shmem.c#L986

LorenzMeier, 2016-04-25:
I'm pretty sure that the list logic there is wrong i.e. its dirty bit is not set or evaluated correctly. Note that there is a "unsaved", "non-default" and "used flag" and those have to be used correctly.

julianoes, 2016-04-25:
The "used flag" is not used on Snapdragon.

I just checked, and you can set a param from default to something in QGC/mavlink and it won't get saved with either: `param save` or `qshell param save`.

julianoes, 2016-04-25:
The problem seems **not** to be the "unsaved" flag here:
https://github.com/PX4/Firmware/blob/master/src/modules/systemlib/param/param_shmem.c#L980

julianoes, 2016-04-25:
@mcharleb this is the param sync issue that I was telling you about earlier.

julianoes, 2016-04-26:
The problem is essentially that the local list of changed parameters on the DSP side is not kept up to date with what is changed from looking at the shared memory.

The local list not in sync is `param_values` in https://github.com/PX4/Firmware/blob/master/src/modules/systemlib/param/param_shmem.c#L971.

If you do `qshell show param PWM_DISARMED` it will update the local list and realize that it has been changed.

@LorenzMeier any advice how to patch this?

LorenzMeier, 2016-04-26:
We need a param update / check API on the DSP side and then register it with the existing always-running daemons (e.g. the HRT) to run regularly.

julianoes, 2016-04-26:
But that would cause a race with the param save because you want to do the update before you do a save.

LorenzMeier, 2016-04-26:
The right way to do this is to establish a lock so they can't race and of course always run the param update before a save.

It actually sounds like this is all what you need to do for now, since the param changed publication makes it to the DSP and we don't need a background process?

julianoes, 2016-04-26:
Ok, right. So I guess, the param update needs to loop through all params and check whether they have been changed. I'm trying to find the param from looking at the changed bit, that would be less cycles.

PR #4369 (2016-04-26): param_shmem: fix bug where params didn't stick
There was the case where a param was changed on the Linux side but the
change did not get saved on the DSP side because the param was not in
the local list of changed params on the DSP side. On save, the param
index is now refreshed, and param_get is called on all params that have
changed. This is a hacky workaround but resolves the problem for now.

Tested on Snapdragon, none of the changes are affecting anything non-Snapdragon.

Resolves #4356.
closed 2016-04-26 by LorenzMeier