PX4/PX4-Autopilot #4364: Sistem hangs on RTL call from RC in single channel mode selection

gianni-carbone, 2016-04-25:
Helo,
after the last firmware update I suffer a serious problem. If I try to switch in Return To Land mode when non GPS signal available the system completely hangs.
The system does normally work when a GPS valid signal is present.
The problem is serious cause if the system is in flight in manual mode with no GPS lock and accidentally the RTL switch is selected the system HANGS whit result of a crash.
firmware: px4fmu-v2_default
Regards GC

LorenzMeier, 2016-04-26:
This was on stable, correct? We have reworked this part of the flight mode assignment on master.

gianni-carbone, 2016-04-26:
It was on the github source at the moment of writing message (after the time of compiling) I don't know if the issue is present on stable version.

commit 2fff2d4e (2016-04-26): Robustify single mode selection. Fixes #4364
closed 2016-04-26 by LorenzMeier