PX4/PX4-Autopilot #4378: EKF2 replay logging is broken.

priseborough, 2016-04-27:
Every 10th or so value of RPL1.t is zero which breaks replay.

devbharat, 2016-04-27:
@tumbili is the fix up somewhere ?

devbharat, 2016-04-27:
I broke the replay message in separate parts for different sensors, had them all in the union except the one for IMU, which was its own struct. That solved it for me, not sure if its the best solution.

devbharat, 2016-04-27:
I finally have ekf2_replay working with vision on master. Something is weird though, I have to overwrite the length of LOG_ATT_MSG in ekf2_replay like this:

    // data message
    int length;
    if(header[2] != LOG_ATT_MSG){
        length = _formats[header[2]].length;
    } else {
        length = 55;  // ATT has length 55, force it!
    }
    if (::read(fd, &data[0], length - 3) != length - 3) {
        PX4_INFO("Done!");
        _task_should_exit = true;
        continue;
    }

as the **_formats[header[2]].length** value for LOG_ATT_MSG is somehow 0, which would otherwise crash the app.

tumbili, 2016-04-28:
@devbharat Try master now without your workaround.

devbharat, 2016-04-28:
Are you re

tumbili, 2016-04-28:
@devbharat Yes, I tried and I get no messages like that.

devbharat, 2016-04-28:
OK, I'll diff the sdlog2 app with master and check then

closed 2016-05-03 by tumbili

zubair123khan, 2016-06-20:
hello guys, i need to replay a log and when i am running replay on it, its throwing me error that no rule to make this file, although i am in right directory. 
please help me out with this..