PX4/PX4-Autopilot #4379: Zubax GNSS v1 and UAVCAN stale mag data

AndreasAntener, 2016-04-27:
UAVCAN with Zubax GNSS v1 on stable (1.2.0) blocks parameter loading in QGC, but that's working on master

After calibration on master the sensor voter reports mag failover while sitting on the ground.

```
nsh> attitude_estimator_q status
attitude_estimator_q: gyro status:
INFO  validator: best: 0, prev best: 0, failsafe: NO (0 events)
INFO  sensor #0, prio: 100, state: OK
INFO    val:  -0.0010, lp:  -0.0011 mean dev:  -0.0000 RMS:   0.0169 conf:   1.0000
INFO    val:   0.0006, lp:   0.0006 mean dev:   0.0000 RMS:   0.0219 conf:   1.0000
INFO    val:  -0.0007, lp:  -0.0017 mean dev:   0.0000 RMS:   0.0551 conf:   1.0000
INFO  sensor #1, prio: 75, state: OK
INFO    val:  -0.0091, lp:   0.0010 mean dev:   0.0000 RMS:   0.0180 conf:   1.0000
INFO    val:  -0.0120, lp:  -0.0011 mean dev:   0.0000 RMS:   0.0226 conf:   1.0000
INFO    val:   0.0091, lp:   0.0039 mean dev:   0.0000 RMS:   0.0554 conf:   1.0000
attitude_estimator_q: accel status:
INFO  validator: best: 0, prev best: 0, failsafe: NO (0 events)
INFO  sensor #0, prio: 100, state: OK
INFO    val:  -0.0282, lp:  -0.0301 mean dev:  -0.0000 RMS:   0.0988 conf:   1.0000
INFO    val:   0.2237, lp:   0.2373 mean dev:   0.0000 RMS:   0.0608 conf:   1.0000
INFO    val:  -9.9446, lp:  -9.9518 mean dev:  -0.0000 RMS:   0.0931 conf:   1.0000
INFO  sensor #1, prio: 75, state: OK
INFO    val:   0.0367, lp:   0.0088 mean dev:  -0.0000 RMS:   0.1013 conf:   1.0000
INFO    val:   0.0964, lp:   0.1497 mean dev:   0.0001 RMS:   0.0653 conf:   1.0000
INFO    val:  -9.8117, lp:  -9.8400 mean dev:  -0.0000 RMS:   0.0978 conf:   1.0000
attitude_estimator_q: mag status:
INFO  validator: best: 1, prev best: 0, failsafe: YES (11 events)
INFO  sensor #0, prio: 50, state: OK
INFO    val:   0.1278, lp:   0.1272 mean dev:   0.0000 RMS:   0.0022 conf:   1.0000
INFO    val:   0.0315, lp:   0.0286 mean dev:   0.0000 RMS:   0.0027 conf:   1.0000
INFO    val:   0.3741, lp:   0.3755 mean dev:  -0.0000 RMS:   0.0014 conf:   1.0000
INFO  sensor #1, prio: 100, state: OK
INFO    val:   0.2588, lp:   0.2582 mean dev:  -0.0000 RMS:   0.0017 conf:   1.0000
INFO    val:   0.1120, lp:   0.1118 mean dev:   0.0000 RMS:   0.0025 conf:   1.0000
INFO    val:   0.5402, lp:   0.5399 mean dev:  -0.0000 RMS:   0.0012 conf:   1.0000
attitude_estimator_q: running
```

The log message I get is MAG#1 stale data

mhkabir, 2016-04-27:
You're running outdated firmware on the GNSS. 

The firmware doesn't publish the mag at a high rate, which causes this. 

There is a parameter on the GNSS side which controls the rate, you need to increase it. The latest firmware has this fixed by default :)

AndreasAntener, 2016-04-27:
@mhkabir thanks for the hint, I could only find this: https://zubax.com/news/zubax-gnss-firmware-update-v20 . Is that the latest binary available or do I have to compile it from source?

Haven't updated it with the binary yet though.

mhkabir, 2016-04-27:
Yeah I think that's the latest. @pavel-kirienko?

pavel-kirienko, 2016-04-28:
@mhkabir thanks for letting me know.

@AndreasAntener there's no need to update the firmware if the current one is working. Please just set the configuration parameter `uavcan.pubp-mag` to 20000. You can do it via UAVCAN or via CLI, in the latter case use the following commands:

```
cfg set uavcan.pubp-mag 20000
cfg save
reset
```

AndreasAntener, 2016-04-28:
@pavel-kirienko thanks, I was able to update the config but having different issues on stable now (sometimes showing 3 mags?).  Couldn't test yet on master (other issues there right now), will try soon.

Memory is a problem, 2 mavlink instances plus uavcan results in a hardfault, also 1 mavlink instance plus uavcan and mag calibration doesn't work on stable.

mhkabir, 2016-05-01:
We fought with the 3 mags problem before. Couldn't find a way to fix it, other than reducing memory consumption. It crops up only when you're low on memory.

I've observed the memory problem in master as well. I'm running 1 mavlink and uavcan, and the calibration routines don't work properly.

closed 2016-05-22 by LorenzMeier