PX4/PX4-Autopilot #4383: Can not start sensors application on Raspberry Pi

nacho-carnicero, 2016-04-27:
Hi, 

I am having some troubles with making the PX4 firmware working on the Raspberry. I cross-compiled the firmware and uploaded it to the board and was able to execute it. Nevertheless, I am having troubles when starting some applications. For example when I start the sensors application by 
```
sensors start
```
I get the following error:
```
WARN  2          ERROR: no barometer found on /dev/baro0 (9) (file ../src/modules/sensors/sensors.cpp line 958)
WARN  FATAL: no accelerometer found: /dev/accel0 (9)
WARN  418        sensor initialization failed (file ../src/modules/sensors/sensors.cpp line 2071)
WARN  539        start failed (file ../src/modules/sensors/sensors.cpp line 2309)
Command 'sensors' failed, returned 1
```

I also get errors when running Mavlink, indeed I get the same errors as in [this page](http://discuss.px4.io/t/px4-on-raspberry-pi-2-problem-with-mavlink-connection/262).

I appreciate any help! Thanks in advance, 

Nacho.

julianoes, 2016-04-28:
Thanks for reporting. I think I know why this is happening. I'll look into it.

julianoes, 2016-04-28:
I tried a fix for this.
@nacho-carnicero can you please test with the branch in #4393?

mhkabir, 2016-04-28:
Err. I think that the DriverFramework wrappers are not running. 

@nacho-carnicero Did you manually start the drivers?

nacho-carnicero, 2016-04-28:
I will try tomorrow morning and let you know, I can't right now. @mhkabir I didn't have the drivers at first included in the makefile, so I couldn't start them manually when I posted this issue. Later on I added the drivers to the makefile and managed to start them manually, but I still had this error when starting the sensors application.

I have tried to run the ./df_imu_test test included in the DriverFramework repo (see [here](https://github.com/PX4/DriverFramework/pull/32)) and wasn't able to run it at first, but I made some modifications and now I get some data from the MPU sensor. I haven't tried if after these modifications the sensor application gets started when running the PX4 firmware. Tomorrow I'll have more time to document better the changes I made and let you know if the modification @julianoes made works.

nacho-carnicero, 2016-04-29:
Ok I tried your modification @julianoes and I got a slight different error:
```
WARN  3          ERROR: no barometer found on /dev/baro0 (9) (file ../src/modules/sensors/sensors.cpp line 958)
WARN  434        FATAL: no ADC found: /dev/adc0 (9) (file ../src/modules/sensors/sensors.cpp line 983)
WARN  477        sensor initialization failed (file ../src/modules/sensors/sensors.cpp line 2073)
WARN  541        start failed (file ../src/modules/sensors/sensors.cpp line 2317)
Command 'sensors' failed, returned 1
```
I think the main problem comes from the fact that the drivers are not included in the build of the firmware, for example when I type help to see the available commands I get this:
```
helpBuiltin Commands:
	attitude_estimator_q
	commander
	dataman
	esc_calib
	fw_att_control
	fw_pos_control_l1
	land_detector
	list_devices
	list_files
	list_tasks
	list_topics
	listener
	mavlink
	mc_att_control
	mc_pos_control
	mixer
	navigator
	param
	perf
	position_estimator_inav
	reboot
	sdlog2
	sensors
	shutdown
	sleep
	uorb
	ver
	vtol_att_control
```
So no MPU driver, or is it automatically initialized when invoking sensors start ?

julianoes, 2016-04-29:
@nacho-carnicero: thanks for trying.

I think you need at least #4397 as well.

I'm not sure if the MPU9250 is already compiled in. @bo-rc should know.

mhkabir, 2016-04-29:
OK, I'm on this already. WIP branch is here : https://github.com/mhkabir/Firmware/tree/rpi2_bringup

Drivers should now be compiled in, but there is still work left to do. @julianoes How do I specify the DriverFramework build type? Some defines are gated with ```__RPI2```, so how do I get them active when building via Firmware?

julianoes, 2016-04-29:
@mhkabir cool, looking forward to that.

Not sure, there should also be defines like `__PX4_RPI2` but they might not exist yet?

mhkabir, 2016-04-29:
@julianoes Yes, don't think so. Will need to fix. Can you give me the locations where those relevant things are done for Eagle?

mhkabir, 2016-04-29:
OK, done. Added definitions for __PX4_RPI2. Will be in my branch soon.

mhkabir, 2016-04-29:
@julianoes OK, there's ```PX4_QURT```, ```PX4_POSIX_EAGLE``` and ```PX4_POSIX_RPI2```.

For consistency, I propose we rename them to just ```PX4_xxx```, like ```PX4_RPI2``` and drop the ```POSIX```

What do you think?

LorenzMeier, 2016-04-29:
No, we're not. The reason is that PX4_POSIX is for operating systems adhering to POSIX standards, while PX4_POSIX_EAGLE is for the POSIX side (= non-QuRT / RTOS) of the EAGLE board. It all makes sense.

mhkabir, 2016-04-29:
OK. Gotcha. Done.

The RPI2 build will export ```__PX4_POSIX_RPI2```.

mhkabir, 2016-04-29:
OK, now how do I get DriverFramework to use the RPI2 define to internally select the bus configs, etc.?

For example here : https://github.com/PX4/DriverFramework/blob/master/framework/include/ImuSensor.hpp#L43

nacho-carnicero, 2016-04-29:
@julianoes I tried and still got the same error, but thanks for the quick reply! Definitely the drivers are not added on the build. 

@mhkabir I think what you're looking for gets solved by adding to the Toolchain-arm-linux-gnueabihf.cmake file this:
```
add_definitions(
	-D__RPI2
	)
```

mhkabir, 2016-04-29:
No that's not what I meant, but got it figured out.

mhkabir, 2016-04-29:
All interested, please see #4404 

:)

julianoes, 2016-07-27:
This works by now.

closed 2016-07-27 by julianoes