PX4/PX4-Autopilot #4386: Drive the submodule checks from the cmake - remove redundant

davids5, 2016-04-27:
remove the module list from here
https://github.com/PX4/Firmware/blob/nuttx_next_cmake_master/Tools/check_submodules.sh#L77-L91

Is is best driven by:

https://github.com/PX4/Firmware/blob/nuttx_next_cmake_master/CMakeLists.txt#L255-L266

and tested in px4_add_git_submodule
https://github.com/PX4/Firmware/blob/nuttx_next_cmake_master/cmake/common/px4_base.cmake#L148-L150

dagar, 2017-04-11:
Actually, I think this needs to be handled by the Makefile. During the cmake configure it won't tolerate its configuration changing, like if one of the submodules contains a CMakeLists.

davids5, 2017-04-11:
@dagar - The only point was there should only be 1 list.  I will leave the magic up to you :)

dagar, 2017-12-13:
Done!

closed 2017-12-13 by dagar