PX4/PX4-Autopilot #4388: Align the way the DriverFramework is pulled into the build with other components.

davids5, 2016-04-27:
Currently there exists DF_TARGET and df_driver_framework is assumed to be need by all target with the special configuration for nuttx.

See https://github.com/PX4/DriverFramework/blob/2eccd3dfb376375e98fcf2c2061c560b37e05af7/framework/src/CMakeLists.txt#L36

the src/lib/DriverFramework should just be added to the cmake/configs/<>.cmake that needed,

This becomes very evident in nuttx_next_cmake and the kludges needed to build bootloaders

See:

 https://github.com/PX4/DriverFramework/blob/2eccd3d_disable_df_for_bootloaders/framework/src/CMakeLists.txt

https://github.com/PX4/Firmware/blob/nuttx_next_cmake_master/cmake/nuttx/px4_impl_nuttx.cmake#L509-L511

and 

https://github.com/PX4/Firmware/blob/nuttx_next_cmake_master/cmake/nuttx/px4_impl_nuttx.cmake#L509-L511

and

mcharleb, 2016-05-02:
Fixed in https://github.com/PX4/Firmware/pull/4415

davids5, 2016-05-17:
Works as advertised!

closed 2016-05-17 by davids5