PX4/PX4-Autopilot #4389: Development Environment on Linux

sgh055, 2016-04-28:
Hi, I have some problem.

I followed Development Environment on Linux(http://dev.px4.io/starting-installing-linux.html#development-environment-on-linux). but When I make px4-v2, I got a error message.
How can i fix that? I installed all of them

This message.
[image]

kd0aij, 2016-04-28:
Are you sure you got everything installed? It looks like you didn't install
gcc.

sgh055, 2016-04-28:
[image]

@kd0aij this??

kd0aij, 2016-04-28:
No, try "arm-none-eabi-gcc --version"

sgh055, 2016-04-28:
[image]
@kd0aij  I try to command, they say "No such file or directory" 

however, file or directory exist.
[image]

Inspirati, 2016-04-28:
What's with the 'ardupilot', are you sure you've got the correct codebase?

mhkabir, 2016-04-28:
Nice catch @Inspirati. Looks like he's trying to build ArduCopter and following the PX4 build steps.

closed 2016-04-28 by sgh055