PX4/PX4-Autopilot #4390: Minimum thrust is zero if mpc_mainthr_min <= 15%

svpcom, 2016-04-28:
On **master** and **beta** branches minimum thrust is 0 if ``mpc_mainthr_min`` <= 15%.
On **stable** branch behavior is normal - thrust is equal to ``mpc_mainthr_min`` if
thrust stick is in down position. Tested with ``make px4fmu-v2_ekf2``  frame is hexarotor in "X" mode.

LorenzMeier, 2016-04-28:
@svpcom I guess I don't understand the question yet completely.

svpcom, 2016-04-28:
In stable branch if I arm copter and stay throttle stick in lower position (zero throttle) PWM output to engines is not PWM_MIN, but mpc_mainthr_min (8% by default). In beta and master branch after engine arm i have PWM_MIN output if mpc_mainthr_min <= 15%  else behavior is the same as in stable.  Also in beta/master branch throttle up/down without takeoff produces messages: takeoff detected -> landing detected.In stable -- "landing detected" appears only after actual takeoff and landing. I've compare code between stable and beta and it seems that the problem in src/modules/land_detector/MulticopterLandDetector.cpp which was patched several months ago

LorenzMeier, 2016-04-29:
@svpcom Can you re-try master? I've improved the land detection there. Its now in a state where I'm happy with it. In particular having 0% PWM on zero manual throttle and the raise once the user throttles up and takeoff is detected. Note that the throttle behaviour also depends on wether you have GPS or not.

svpcom, 2016-04-29:
With latest master i've got following after bootup:
[image]

qgroundcontrol is a latest trunk (works normally with px4 stable)

LorenzMeier, 2016-04-29:
Your status bar also looks wrong. Do you have a 2nd Pixhawk on which you can reset the params? I'm wondering if your boot aborts due to some local settings.

svpcom, 2016-04-29:
After pixhawk reboot i've got infinite loop of: 
[image]

I've found the only way to stop it is to define some waypoints, upload them to vehicle and reboot pixhawk. After that it shown on boot: ``Critical: Operation timeout``. When I attempt to alter waypoints and upload them to pixhawk i've got endless ``No active transfer`` again. I've tried to erase mtd via ``mtd erase`` -- no success. My px4 settings:
[settings.zip](https://github.com/PX4/Firmware/files/243126/settings.zip)

LorenzMeier, 2016-04-29:
Can you rebuild master and try again (or flash through QGC, but wait 10 mins before you do)? I just fixed an issue in the MAVLink app.

svpcom, 2016-04-29:
Ok. I'll try. Toolbar is half empty because i've disconnect a pixhawk to make a screenshot. What is a preferred way to reset settings to factory defaults? I've tried Parameters->Tools->Reset all to defaults. It clears all settings (including sensors calibrations) but continue show endless ``No active request``.
Also what build target i should use: px4fmu-v2_ekf2 or px4fmu-v2_default ? Now I use ekf2

svpcom, 2016-04-29:
With latest master 796af52df26599fc749fa7a4ab446c2640e38662 behavior is the same:
```
[svpcom@localhost mycopter]$ ./mavproxy/bin/mavproxy.py 
WARNING: You should uninstall ModemManager as it conflicts with APM and Pixhawk
Auto-detected serial ports are:
/dev/serial/by-id/usb-3D_Robotics_PX4_FMU_v2.x_0-if00
Connecting to /dev/serial/by-id/usb-3D_Robotics_PX4_FMU_v2.x_0-if00
Connect /dev/serial/by-id/usb-3D_Robotics_PX4_FMU_v2.x_0-if00 source_system=255
Log Directory: 
Telemetry log: mav.tlog
MAV> Waiting for heartbeat from /dev/serial/by-id/usb-3D_Robotics_PX4_FMU_v2.x_0-if00
online system 1
Mode(65536)> Mode Mode(65536)
APM: WPM: REJ. CMD: partner id mismatch
fence breach

Mode(65536)> APM: WPM: REJ. CMD: partner id mismatch
APM: WPM: REJ. CMD: partner id mismatch
Received 386 parameters
Saved 387 parameters to mav.parm
APM: WPM: REJ. CMD: partner id mismatch
APM: WPM: REJ. CMD: partner id mismatch
APM: WPM: REJ. CMD: partner id mismatch
APM: WPM: REJ. CMD: partner id mismatch
APM: WPM: REJ. CMD: partner id mismatch
APM: WPM: REJ. CMD: partner id mismatch
APM: WPM: REJ. CMD: partner id mismatch
APM: WPM: REJ. CMD: partner id mismatch
APM: WPM: REJ. CMD: partner id mismatch
```

I've made build as:
```
make distclean
make px4fmu-v2_default upload
```

Reset all settings also no help. In SITL mode there are no such messages.
My toolchain is: gcc-arm-none-eabi-4_8-2014q3.
If you need a boot log i can connect serial cable and capture it.

svpcom, 2016-04-29:
Boot log:
```
nsh> reboot
sercon: Registering CDC/ACM serial driver
sercon: Successfully registered the CDC/ACM serial driver
[i] microSD mounted: /fs/microsd 
param: selected parameter default file /fs/mtd_params
[param] Loaded: /fs/mtd_params 
rgbled on I2C bus 2 at 0x55 (bus: 100 KHz, max: 100 KHz)
 1:   SYS_USE_IO: match
 6001: + SYS_AUTOSTART: match
px4io: CRCs match
dataman: Power on restart, data manager file '/fs/microsd/dataman' size is 103090 bytes
MS5611_SPI on SPI bus 1 at 3 (20000 KHz)
bst: no devices found
adc init done
ver hwcmp match: PX4FMU_V2
HMC5883_I2C on I2C bus 1 at 0x1e (bus: 100 KHz, max: 400 KHz)
lis3mdl: no device on bus 2
LIS3MDL bad ID: 3chmc5883: no device on bus 1
mpu6000: no device on this bus
MPU6000 on SPI bus 1 at 4 (1000 KHz)
L3GD20 on SPI bus 1 at 1 (11000 KHz)
LSM303D on SPI bus 1 at 2 (11000 KHz)
meas_airspeed: no MS4525 airspeed sensor connected
ets_airspeed: no ETS airspeed sensor connected
ets_airspeed: no ETS airspeed sensor connected
nsh: sf10a: command not found
px4io default PWM output device
mavlink_if0: mode: 0, data rate: 1200 B/s on /dev/ttyS1 @ 57600B
 157600:   SYS_COMPANION: match
mavlink_if1: mode: 3, data rate: 1000 B/s on /dev/ttyS2 @ 57600B
ver hwcmp match: PX4FMU_V2
px4flow [166:100]
px4flow: scanning I2C buses for device..
mavlink_if2: mode: 5, data rate: 800000 B/s on /dev/ttyACM0 @ 57600B
WARN  log buffer size: 12288 bytes
[i] MULTICOPTER 
[i] Mixer: /etc/mixers/hexa_x.main.mix on /dev/pwm_output0 
[i] Mixer: /etc/mixers/pass.aux.mix on /dev/pwm_output1 
 0:   SYS_MC_EST_GROUP: match

NuttShell (NSH)
nsh> mavlink_if1: Disabling hardware flow control
mavlink_if1: stream SYS_STATUS on device /dev/ttyS2 enabled with rate 100.000 Hz
mavlink_if1: stream SYSTEM_TIME on device /dev/ttyS2 enabled with rate 100.000 Hz
mavlink_if1: stream SYS_STATUS on device /dev/ttyS2 enabled with rate 100.000 Hz
mavlink_if1: stream SYSTEM_TIME on device /dev/ttyS2 enabled with rate 100.000 Hz
mavlink_if1: stream SYS_STATUS on device /dev/ttyS2 enabled with rate 100.000 Hz
mavlink_if1: stream SYSTEM_TIME on device /dev/ttyS2 enabled with rate 100.000 Hz
mavlink_if0: Disabling hardware flow control
WARN  Not ready to fly: Sensors not set up correctly
WARN  Not ready to fly: Sensors not set up correctly
```

svpcom, 2016-04-29:
I've found the root of issues - they are triggered by my OSD (PlayuavOSD) which is connected via TELEM2 port. But what is changed in px4 mavlink protocol since stable release that could trigger such behavior?

svpcom, 2016-04-29:
I've tested land detector fixes - all works prefectly! Will it be ported to beta?

LorenzMeier, 2016-05-01:
Yes, its on beta now. I also pushed improvements to it in an attempt trying to solve the messaging issue you had - did it fix that?

svpcom, 2016-05-01:
I've already fixed messaging issue by disable waypoints sync in OSD code. I'll debug it later

closed 2016-05-01 by LorenzMeier