PX4/PX4-Autopilot #4395: Support mavlink over UART with Snapdragon

julianoes, 2016-04-29:
It would be nice to be able to run mavlink with a 3DR radio connected over serial to the Snapdragon.

jywilson, 2016-05-06:
@julianoes There is a file called blsp.config that will allow you to configure 2-wire UART access to a particular BLSP.  This can be important when you are trying to use a single DF-13 connector for multiple types of output, as with UART/I2C.

Are you familiar with this file?  I will add to the Snapdragon markdown to document the format of the file.

pzoom, 2016-08-03:
@jywilson could you send me a link to the snapdragon markdown document you reference above? I'm interested in the blsp.config file and what the options are for connecting I2C to ESC (specifically the VESC http://vedder.se/2015/01/vesc-open-source-esc/).

I'm also interested in configuring one of the BLSPs as a UART for connection to QGroundcontrol via a higher power radio, e.g  http://www.microhardcorp.com/n920.php which acts as a transparent RS232 cable.

pzoom, 2016-08-04:
@julianoes do you know if there's documentation for the blsp.config file and where I can access it? Thanks!

julianoes, 2016-08-04:
@pzoom I don't know sorry.

closed 2018-02-21 by PX4BuildBot

TSC21, 2018-02-21:
@julianoes @ChristophTobler is this still relevant?

julianoes, 2018-02-22:
Not for me. I am reluctant with all the Snappy issues since I the board is end-of-lifed soon anyway.