PX4/PX4-Autopilot #4408: FrSky D type telemetry broken in beta and master

dzett, 2016-04-30:
Hi,

my Pixracer has been sitting around a few weeks due to other stuff and I had not updated it. When I flashed today's master (also tried beta), I noticed that I am not getting any telemetry data on my FrSky D-type radio. When I went back to a previously saved master build downloaded on March 26, telemetry was working fine again.

Is this a known issue? I noticed some cleanup has taken place in the FrSky telemetry code, maybe an issue there?

Let me know how I can help with analysis.

Best regards,
Daniel

thedevleon, 2016-04-30:
Hey @dzett, thanks for mentioning this. Seems like I made a small mistake when I edited some of the frsky telemetry code, while I'm currently implementing further changes to the sPort Telemetry Code.
I have created a PR [here](https://github.com/PX4/Firmware/pull/4409) which should fix the issue. I don't have a D-Type receiver on my, so could you test this please and report back?

dzett, 2016-04-30:
Hi @thedevleon, thanks for the quick response. I pulled your fix #4409 and it is working fine for me, telemetry is back.

LorenzMeier, 2016-05-01:
Pushed to master and beta, thanks!

closed 2016-05-01 by LorenzMeier

spinorkit, 2017-12-30:
FrSky D-type telemetry is broken  for my PixHawk v2 in PX4 version 1.7.2 (current stable).
It works if I go back to v1.1.2 and use an extras.txt file with:
frsky_telemetry start -d /dev/ttyS6

Any ideas on how to get it to work with the latest firmware. Does it still need an extras .txt file with the PixHawk v2?

dagar, 2017-12-30:
@spinorkit please open a new issue. Did you try the extras.txt on v1.7.2? Did you get any error messages? Are you able to connect a system console? https://dev.px4.io/en/debug/system_console.html