PX4/PX4-Autopilot #4425: QGC/PX4 not communicating correctly

srees, 2016-05-03:
Behavior is similar to that caused on a linux system when e2fd2f4 was implemented, same as #3707, has come up again. 

Fresh build of PX4 firmware on Ubuntu VM, latest QGC running on Mac host, different IP addresses.
QGC detects the launch of 'make posix gazebo' but immediately loses communication with it. Telem logs from the make command indicate connection is still active (closing/relaunching QGC is reflected with 'data link#0 lost/regained), but nothing is coming through to QGC. QGC reports that 'vehicle did not respond to request for parameters'.

LorenzMeier, 2016-05-03:
This is actually intentional since it caused havoc in local networks. You would need to remove the -t parameter here: https://github.com/PX4/Firmware/blob/master/posix-configs/SITL/init/rcS_gazebo_iris#L54

I don't have a one-size-fits-all solution yet.

srees, 2016-05-05:
That's helpful to know, thanks. Is there documentation somewhere of what the command switches (parameters) do for mavlink? I can't seem to find that.

srees, 2016-05-05:
Nevermind, found usage at the end of mavlink_main.cpp

warnx("usage: mavlink {start|stop|stream} [-d device] [-u network_port] [-o remote_port] [-t partner_ip] [-b baudrate]\n\t[-r rate][-m mode] [-s stream] [-f] [-p] [-v] [-w] [-x]");

srees, 2016-05-05:
So if I understand this right, QGC will work fine if it is running on the same host as the SITL simulation is running?

LorenzMeier, 2016-05-05:
That's correct.

closed 2016-05-22 by LorenzMeier