PX4/PX4-Autopilot #4428: Simulator: add flag to disable battery simulation in SITL

julianoes, 2016-05-03:
It would be nice to have a command line flag for `simulator` which disabled the battery simulation for SITL. This allows for testing of anything that is not battery (or failsafe) related.

LorenzMeier, 2016-05-03:
The simulation will not actually discharge to a failsafe level. If you like we can set the warning level low enough (just altering the param value) so it doesn't trigger the warning. No need to remove the battery drain.

julianoes, 2016-05-04:
I think it makes sense to have options for the following cases:
1. Normal drain where failsafe is triggered.
2. Discharge but no triggering of failsafe (like now).
3. No discharge, for simulation tests going (almost) infinitely.

LorenzMeier, 2016-12-26:
Closing as stale, seemed to be workable for folks.

julianoes, 2019-07-01:
This was still on my personal todo list :smile:

hamishwillee, 2019-07-02:
When this is done, let me know so I can document it. I think it would be good to be able to simulate the battery failsafe. If this is already possible, what parameter needs to be set?

julianoes, 2019-07-02:
We want to be able to properly simulate all sorts of failsafes and failsafe combinations in SITL and in real flights, and this is just a part of it.

mhkabir, 2019-07-02:
Saw some somewhat related progress towards this here: https://github.com/PX4/Firmware/tree/feature/sim-battery-parameter

hamishwillee, 2019-10-01:
@julianoes The tree in https://github.com/PX4/Firmware/issues/4428#issuecomment-507524004 disappeared. I presume we still want something for this?

julianoes, 2020-03-01:
This is doable with params `SIM_BAT_DRAIN` and `SIM_BAT_MIN_PCT`.

closed 2020-03-01 by julianoes

AlexWUrobot, 2023-02-03:
" [health_and_arming_checks] Preflight Fail: Battery unhealthy"
The same issue happened when running on the simulation.
I don't know how to solve this issue on the simulation.
Thanks a lot
ref:https://discuss.px4.io/t/preflight-fail-battery-unhealthy/30678