PX4/PX4-Autopilot #4431: Calibration PX4Flow

quentin1002, 2016-05-04:
Hi everybody, 

I followed this [tutorial](http://ardupilot.org/copter/docs/common-px4flow-overview.html) in order to calibrate my PX4Flow, used along with a pixhawk PX4 board. Neverless the consistency check is not giving me a relevant result. Here is what I get performing the x-gyro test. In white is the HIGH_RES_IMU_xgyro and in pink is the OPTICAL_FLOW_RAD_Integrated_xgyro. As you can see there is a scaling effect that I can't solve...
[image]

Thanks in advance for your help,
Quentin

proficnc, 2016-05-04:
Are you doing this in bright sunlight?  Low light will give bad results.

quentin1002, 2016-05-04:
Yes I am doing with good light, though interior, and the image quality is around 240

proficnc, 2016-05-04:
If indoor, ensure your lights are incandescent

quentin1002, 2016-05-04:
in your opinion, light is the only reason? Because I just tried to add an additional light (no effect), move outside of the office with direct sun light (no effet). I also tried to vary the pattern of the floor and the height.... I still have the IMU pic 3 times higher than the OF's.

proficnc, 2016-05-04:
No, just worth checking the basics.

What calibration settings are you using?

quentin1002, 2016-05-04:
What do you mean by settings? I am using QGroundControl V2.9.1. 

The PX4 was calibrated following the 4 steps of QGC and regarding the px4flow is used the video mode only to focus the image. My board is used at around 1m from the ground....

proficnc, 2016-05-04:
Which instructions are you following?

quentin1002, 2016-05-04:
The "official" [one](https://pixhawk.org/modules/px4flow)

proficnc, 2016-05-04:
Ok, I'm not familiar with using flow with px4 flight, but setting up for Ardupilot is described well here.

http://ardupilot.org/copter/docs/common-px4flow-overview.html

LorenzMeier, 2016-05-04:
Please use the official instructions from the hardware site, not any 3rd party instructions which are often stale wrt to the official release firmware. The reason you're confused is that you need to read up on the message spec: the sensor is sending the integrated velocity, not angular rate.

closed 2016-05-04 by LorenzMeier