PX4/PX4-Autopilot #4434: New parameter generation does not work for Snapdragon

julianoes, 2016-05-04:
The Snapdragon build contain `qurt_eagle_default` for the DSP side and `posix_eagle_default` for the Linux side. Not all parameters are built/included on both sides.

Therefore the new param generation does not work on Snapdragon because the two sides are not in sync and refer to different params for a specific ID which really messes things up. (E.g. when trying to do RC calibration today, we accidentally changed MPC control gains which didn't fly very nicely :( )

FYI: @davids5

mcharleb, 2016-05-04:
Could all the params be generated and then pruned to the set that is actually used so the IDs remain the same?

davids5, 2016-05-04:
@julianoes @mcharleb 

I think there are 2 choices:

1) Do not cull the param for a Snapdragon build (needed for memory constrained targets) or 
2)  Add a some sort of relationship that will pull in the union of the 2 files.

julianoes, 2016-05-04:
I would like 1) for the short term, if that's a simple thing to do.

davids5, 2016-05-04:
Let me take a look...

LorenzMeier, 2016-05-04:
@davids5 Thanks! Since this blocks Snapdragon on master a quick fix would be appreciated.

davids5, 2016-05-04:
Sorry for the delay....See https://github.com/PX4/Firmware/pull/4435

julianoes, 2016-05-05:
1) is in master.

Let's leave this issue open for a potential 2).

davids5, 2016-05-05:
@julianoes 

For 2) Would you be able to provide me with the way I would associate the files?  If we can not guarantee that the binary files loaded on system are tied in revision and association then the number 2 option is not a good one.

Also keep in mind that the delta in size  with a px4fmu-v2_default build with ALL parameters is about +3500 bytes. 

When it is your first 3.5K , it is not a big deal, but when it is your last 3.5K it is a show stopper :)

mcharleb, 2016-05-05:
If the configs were organized as:

Firmware/cmake/configs/${BOARD}/${CONFIG}/${OS}.cmake then all the files in the subdir would be used:

Firmware/cmake/configs/eagle/default/posix.cmake
Firmware/cmake/configs/eagle/default/qurt.cmake

Firmware/cmake/configs/px4-stm32f4discovery/default/nuttx.cmake

The union of all files in the subdir would be used to determine the set of required parameters.

davids5, 2016-05-22:
I think the solution that is in place will hold us until we run out of free space 
@julianoes Can we close this and reopen in future if need be?

julianoes, 2016-06-08:
@davids5 Yer it's good enough for now, let's close. Sorry for the long silence on this.

closed 2016-06-08 by julianoes