PX4/PX4-Autopilot #4438: FLAPS: Strange behavior...

tubeme, 2016-05-04:
Hi guys,

I prepare a plane with flaps and retracts for auto take off and landing.

So I stumble on the following problem.

In Stabalized flight or any assisted flight the flaps extension becomes twice as small.

The moment I switch to Manual the flaps extend full way. then flip back to stabalized and they retract half way. I use 3 position switch so I have two positions of the flap. The same two positions I get in the assisted modes but half scaled. 

Is this on purpose or is it a bug?

LorenzMeier, 2016-05-05:
@AndreasAntener and @SimonWilks Can comment in more detail, but I suspect this is design intent to have the flaps extension scale properly with airspeed. It would be easy to change if we conclude this is not a good idea.

LorenzMeier, 2016-05-05:
At zero airspeed the reading is considered invalid and they extend for the trim speed. If you would create airflow with a hair dryer I expect they will extend further.

tubeme, 2016-05-05:
No they don't... Probably something left in the code.

Also there are some considerations for this design. It means that if I have a faulty Airspeed and 0 reading then I will not be able to fully use my flaps during stabalized or gps landing. Dynamic flaps dependent on speed is very nice idea, but I think it should work the opposite way, ie. The lesser the speed the more flaps extention. At 0 speed we get fully extended slaps, this is just in case of airspeed failure. 

There is another thing that could be considered for using flaps. All the planes I've flown and seen have the following problem. When using flaps they have the tendency to pitch-up, The more the speed the more the pitch up sometimes even making a loop. So we program a little pitch down at the same time as the flaps extend. I could do this in the mixer probably but it would be nice to have a parameter that you can command pitch down. I suppose that if the plane is in stabilized it wont do this pitch up on flaps because the computer will compensate. But in acro this trim is necessary. When I try the functioning in couple of days will report.

Thanks Lorenz for the answer.

AndreasAntener, 2016-05-06:
Airspeed scaling is not taken into account for the flaps and they should be the same in all modes (except for auto where they are turned on/off by the position controller).
What is your airframe config? Are these actual flaps or flaperons (mixed/combined with ailerons)?

tubeme, 2016-05-06:
@AndreasAntener OK I understand. So there is a problem because they should be same for 0 airspeed as well!

Airframe is standard AETR airplane. Frame is Durafly T28 with real flaps (not flaperons). I have a custom mixer AETR.main.mix

CH6
FLAPS
------------
M: 1
O:      5000  10000     0 -10000  10000
S: 0 4  10000  10000      0 -10000  10000

Ch7
Landing Gear
-------------
M: 1
O:      10000  10000      0 -10000  10000
S: 3 5  10000  10000      0 -10000  10000

dagar, 2016-11-08:
@tubeme is this still an issue? The flaps code has been updated a bit.

https://github.com/PX4/Firmware/blob/master/src/modules/fw_att_control/fw_att_control_main.cpp#L879

tubeme, 2016-11-09:
I will try it in the coming weeks with another plane I'm preparing for this tests. If it is not corrected this is HUGE issue for sure. Especially for auto take off and landing on runway. I will try it and report thanks.

dagar, 2016-11-10:
We can definitely fix it, especially if you're able to easily reproduce the issue and test potential fixes. Flaps aren't currently part of FW takeoff, but that's an easy change we can talk about once this issue is resolved.

tubeme, 2017-07-07:
Follow up

In 1.6.3 the FLAPS are operational but there are still some concerns about the flaps behavior.

In manual modes the flaps extends full way.

In guided modes ie. Altitude and Position the flaps extend half way.

Also there is very nice touch in Pos/Alt modes that the flaps does not extend full speed but rather we see slow movement from point to point. This is very nice because fully extending flaps right away is not good idea in the airplanes. So this could be applied to the manual modes as well. 

There might be a parameter to allow us to choose how long it will take to extend the flaps. 1,2,3 seconds etc.

I agree that it has logic to extend the Flaps a little less on the safe side for Position and Altitude modes but half way is too much. I could not find a way to make it fully extend in guided modes.

tubeme, 2017-07-07:
Created Follow up in #7548

closed 2017-07-07 by tubeme