PX4/PX4-Autopilot #4443: Target `make px4fmu-v2_ekf2` does not run ekf2

julianoes, 2016-05-05:
I built and uploaded using `make px4fmu-v2_ekf2 upload`, so I would expect ekf2 to be running. 
However, looking at `top`, that doesn't seem to be the case:

```
Processes: 24 total, 3 running, 21 sleeping
CPU usage: 30.90% tasks, 0.52% sched, 68.58% idle
Uptime: 26.481s total, 16.931s idle

 PID COMMAND                   CPU(ms) CPU(%)  USED/STACK PRIO(BASE) STATE 
   0 Idle Task                   16931 68.582     0/    0   0 (  0)  READY 
   1 hpwork                        621  2.686   612/ 1592 192 (192)  w:sig 
   2 lpwork                         89  0.298   572/ 1592  50 ( 50)  w:sig 
   3 init                         1502  0.000  1444/ 2496 100 (100)  w:sem 
 262 top                           154  3.507  1204/ 1696 100 (100)  RUN   
  90 gps                            27  0.074   796/ 1192 220 (220)  w:sem 
  92 dataman                        22  0.000   652/ 1192  90 ( 90)  w:sem 
 117 sensors                       644  2.388  1664/ 1992 250 (250)  w:sem 
 119 commander                     213  0.447  3436/ 3592 140 (140)  w:sig 
 120 commander_low_prio            488  0.000   780/ 2992  50 ( 50)  w:sem 
 125 px4io                         763  3.059   908/ 1496 240 (240)  w:sem 
 135 mavlink_if0                   241  0.895  2324/ 2896 100 (100)  w:sig 
 136 mavlink_rcv_if0                 1  0.000   820/ 2096 175 (175)  w:sem 
 143 mavlink_if1                   183  0.820  2324/ 2896 100 (100)  w:sig 
 144 mavlink_rcv_if1                 1  0.000   796/ 2096 175 (175)  w:sem 
 172 mavlink_if2                  1323  5.671  2332/ 2896 100 (100)  w:sig 
 173 mavlink_rcv_if2                 4  0.000  1032/ 2096 175 (175)  w:sem 
 179 sdlog2                         68  0.000  2764/ 3392 177 (177)  w:sem 
 180 sdlog2_writer                  37  0.000  1044/ 2040  95 ( 95)  w:sem 
 223 attitude_estimator_q         1191  5.223  2012/ 2496 250 (250)  w:sem 
 229 position_estimator_inav       583  2.686  4940/ 5296 250 (250)  w:sem 
 238 mc_att_control                550  2.462  1316/ 1496 250 (250)  w:sem 
 242 mc_pos_control                138  0.671  1220/ 1896 250 (250)  w:sem 
 251 navigator                       2  0.000   900/ 1496 105 (105)  w:sem
```

julianoes, 2016-05-05:
According to http://dev.px4.io/advanced-switching_state_estimators.html#how-to-enable-different-estimators, the following is needed.

```
param set SYS_MC_EST_GROUP 2
param save
```

I think this should be easier.

LorenzMeier, 2016-05-05:
We can't really avoid it as its a workaround for too low flash memory. On FMUv4 none of the special compile targets is required, you just set the parameter and you're good.

LorenzMeier, 2016-05-05:
The special target can be removed once EKF1 has been replaced for fixed wing with EKF2. @CarlOlsson is working on that.

julianoes, 2016-05-05:
I understand.

closed 2016-05-05 by julianoes