PX4/PX4-Autopilot #4461: mavlink debug compile error

Michvw, 2016-05-06:
When i uncomment the px4_mavlink_debug out of my config file, it compiles fine. However when i use the same with the same imports in my own module i'm getting compile errors:

> error: C99 designator 'key' outside aggregate initializer
>   struct debug_key_value_s dbg = { .key = "vel", .value = 0.0f };
>                                                                ^
> compilation terminated due to -Wfatal-errors.

As a result of this error i changed the example code to:

```
struct debug_key_value_s dbg;
orb_advert_t pub_dbg = orb_advertise(ORB_ID(debug_key_value), &dbg);
dbg.value = 0.0f;
dbg.key= 'test';
orb_publish(ORB_ID(debug_key_value), pub_dbg, &dbg);
```
But this gives compile error:

> error: incompatible types in assignment of 'int' to 'int8_t [10] {aka signed char [10]}'
>   dbg.key= 'test';
>          ^
> compilation terminated due to -Wfatal-errors.

Does someone have a clue of what the problem is here?

LorenzMeier, 2016-05-22:
You are facing normal C vs C++ portability issues.

closed 2016-05-22 by LorenzMeier

stribuda, 2019-02-25:
> When i uncomment the px4_mavlink_debug out of my config file, it compiles fine. However when i use the same code with the same imports in my own module i'm getting compile errors:
> 
> > error: C99 designator 'key' outside aggregate initializer
> > struct debug_key_value_s dbg = { .key = "vel", .value = 0.0f };
> > ^
> > compilation terminated due to -Wfatal-errors.
> 
> As a result of this error i changed the example code to:
> 
> ```
> struct debug_key_value_s dbg;
> orb_advert_t pub_dbg = orb_advertise(ORB_ID(debug_key_value), &dbg);
> dbg.value = 0.0f;
> dbg.key= 'test';
> orb_publish(ORB_ID(debug_key_value), pub_dbg, &dbg);
> ```
> But this gives compile error:
> 
> > error: incompatible types in assignment of 'int' to 'int8_t [10] {aka signed char [10]}'
> > dbg.key= 'test';
> > ^
> > compilation terminated due to -Wfatal-errors.
> 
> Does someone have a clue of what the problem is here?

Did you found a solution please?
Thanks!

Chraebi, 2019-06-02:
Dear all,
I'm encountering the same problem, may someone help?

Or is there another way to debug?
Kind regards

hamishwillee, 2019-06-02:
@Chraebi Please don't append to old closed issues. If you're after general help post on the discussion boards. If you clearly have a bug then create your own new issue including all the information about what you're seeing, the platforms you're using, how you're testing. If you are fairly sure that the problem is the same, link to this old issue.

TimTom68, 2019-06-19:
I'm encountering the same problem too, which is a shame since i just followed the tutorial.  I feel like a proper solution should be posted since other people will follow the tutorial too and inevitably end up here

Chraebi, 2019-06-19:
Dear TimTom
Please have a look at https://discuss.px4.io/t/debugging-in-sih/11243/7?u=chraebipk
I hope this solves your issue.
Kind regards
Chraebi