PX4/PX4-Autopilot #4466: VTOL transition switch broken in current master

sanderux, 2016-05-06:
It appears to be related to the radio settings
Transition switch is ignored (quad spins up with throttle when switch on)
Also, the pusher ESC does not initialize, or only does so after changing the VT_IDLE_PWM_MC, it does not seem to matter to what direction to change this.

@AndreasAntener @tumbili @LorenzMeier

LorenzMeier, 2016-05-06:
Make sure to run "make clean" and compile master fresh before uploading

sanderux, 2016-05-06:
Was done, also been able to reproduce om master precompiled through QGC
Additionally the wing servo's do not respond so i'm quite sure this
https://github.com/PX4/Firmware/blob/master/src/modules/vtol_att_control/vtol_att_control_main.cpp#L466
returns false. When reverting to older firmware (28 march) the problem disappears

LorenzMeier, 2016-05-06:
Anyone feeling like adding a proper transition switch? We could squeeze that in for 1.3 and also add QGC config support. Would be nice to have this less obscure.

sanderux, 2016-05-06:
Last working commit i was able to compile: https://github.com/PX4/Firmware/commit/ebacd4c1de5a0231ad9a4cddcc3759a5389835ea

First failure commit i was able to compile:
https://github.com/PX4/Firmware/commit/4bdfb8a10775ab2f8feeade1c622de259234052c

sanderux, 2016-05-06:
It seems to have broken on Apr 11 after @kd0aij's commit. I run into a lot of compile errors when attempting to compile half-way. 
@LorenzMeier can you have a look?

tubeme, 2016-05-06:
We found out this just yesterday! Making a new machine with new Pix flashing Master, loading our perfect settings set and DISASTER. On the bench test, no transition. Also programming acro switch has a very strange behavior in the QGC Flight Mode Setup Page. When we switch to stabilize the QGC reports it in the status bar the voice pronounce it correct but in the Flight Mode setup page it show we are in Acro and vice versa.

Also we could not take off with the quad. The quad seemed like it is in total disorder, even worse than acro. The VTOL just flipped and flipped and flipped... no take off. We had to move back to Stable... But it lacks a lot of the new things.

sanderux, 2016-05-07:
It seems the switches config broke. I will try to dig deeper to find out where it broke

commit 1815b47f (2016-05-07): Add reserved type, fixes #4466 (#4476)

* Add reserved type, fixes #4466

* Additional reserved vtol types added to is_vtol
closed 2016-05-07 by LorenzMeier