PX4/PX4-Autopilot #4477: 1.3 beta three step compass calibration wrong data.

gregswf, 2016-05-07:
I've tried several times to calibrate the compass with the three step process in the 1.3 beta, every time I get the heading off by about 30deg.
I've calibrated the compass using the stable firmware and the six point process and it works every time.

LorenzMeier, 2016-05-07:
Calibration method (looking good):
https://youtu.be/OOKthp_VciE

Good compass cal params:
https://files.gitter.im/PX4/Firmware/jj8F/px4_stable.params

Bad compass cal params:
https://files.gitter.im/PX4/Firmware/lDoq/px4_13_beta.params

LorenzMeier, 2016-05-07:
If my latest changes on master fix it for Greg we can stick to 3 positions, if not we'll return to 6 positions.

gregswf, 2016-05-08:
I've done some tests and this method gives a good result
1.Hold the craft horizontaly. turn round the yaw axis.
2.Hold the craft nose down. turn round the roll axis.
3.Hold the craft nose down. Turn round the pitch axis.
This was suggested by Lorenz, I'm testing it to find out how reliable it is, so far it works very well.

closed 2016-05-19 by LorenzMeier