PX4/PX4-Autopilot #4480: Separate stab/acro gains

tumbili, 2016-05-08:
We need another set of rate gains for the acro mode. The bandwidth of the rate loop in acro mode is lower than when in stabilized mode and as a result the optimal acro gains are unflyable in stabilized mode.
Log showing good acro performance on a 250 quad (roll axis is a bit too tight, small oscillations visible and audible):
http://logs.uaventure.com/view/Dx472FtZUDkAd88Q93PhAS

tumbili, 2016-05-08:
@kd0aij FYI

kd0aij, 2016-05-08:
Have you looked at any of my logs? Which portion of your log is ACRO mode?

tumbili, 2016-05-08:
@kd0aij Yeah I looked at that before. The entire flight is acro, I cannot switch to stabilize otherwise my quad would blow up.

tumbili, 2016-05-08:
@kd0aij I could back down the the attitude p gains but I remember that it would still get unstable when I was testing your gains. BTW, are you still constraining the maximum gyro bias?

kd0aij, 2016-05-08:
which param was that? I'll check

tumbili, 2016-05-08:
attitude estimator q has a parameter which limits the max gyro bias, that's the one I mean.

kd0aij, 2016-05-08:
so my bias weight is 1/10 of the default setting

tumbili, 2016-05-08:
@kd0aij Actually stabilized was also "stable" when I went to test it. I guess there is a ground effect which makes you think that it's not going to be stable but once it's in the air it's ok.
However, it's very aggressive, e.g. if you let the stick pop back to the center you hear the oscillations before they decay. Probably the user could just use the slider to determine what behaviour he wants, e.g. soft non-overshooting or fast with overshoot.

tumbili, 2016-05-08:
@kd0aij Did you observe the same behaviour?

tumbili, 2016-05-08:
@kd0aij BTW, I mounted the Pixracer on foam last Friday, that's the only difference I can think of to before.

kd0aij, 2016-05-08:
yes, in ground effect I see a tendency to oscillate, but that's gone by about 10cm AGL
And yes, it's aggressive in stabilize mode, but not in acro; in acro it just tracks your rate inputs very closely.

julianoes, 2016-05-09:
I disagree that this is needed. Why should the acro gains not work in stabilized mode?
I can only think of the case where the stabilized gains are too high and "ask" the rate controller too much.

LorenzMeier, 2016-05-10:
Not necessary after more testing.

closed 2016-05-10 by LorenzMeier

garlinplus, 2016-06-15:
hi tumbili! I tune the pid of my drone.But no matter how bigger I increase p in acro mode on angular rate loop,it doesn't go to oscillate even if I increase it to 2000.