PX4/PX4-Autopilot #4486: External compass treated as internal first time calibrated

jgoppert, 2016-05-09:
If you reboot and re-calibrate again it will calibrate as external. Seems to be recent.

jgoppert, 2016-05-09:
```
1	1	CAL_MAG0_ID	73225	6
1	1	CAL_MAG0_ROT	-1	6
1	1	CAL_MAG0_XOFF	0.342069178819656372	9
1	1	CAL_MAG0_XSCALE	0.980295240879058838	9
1	1	CAL_MAG0_YOFF	-0.055439420044422150	9
1	1	CAL_MAG0_YSCALE	1.090277552604675293	9
1	1	CAL_MAG0_ZOFF	0.516482889652252197	9
1	1	CAL_MAG0_ZSCALE	1.054827451705932617	9
1	1	CAL_MAG1_ID	131594	6
1	1	CAL_MAG1_ROT	-1	6
1	1	CAL_MAG1_XOFF	0.048744492232799530	9
1	1	CAL_MAG1_XSCALE	1.000000000000000000	9
1	1	CAL_MAG1_YOFF	-0.173162430524826050	9
1	1	CAL_MAG1_YSCALE	1.000000000000000000	9
1	1	CAL_MAG1_ZOFF	-0.370307326316833496	9
1	1	CAL_MAG1_ZSCALE	1.000000000000000000	9
1	1	CAL_MAG2_ID	0	6
1	1	CAL_MAG2_ROT	-1	6
1	1	CAL_MAG_PRIME	73225	6
1	1	CAL_MAG_SIDES	63	6
```

Also not clear in latest qgc binary that we need to calibrate the other 3 sides for mag, or how to default the setting to just 3 sides. Not sure if this is fixed on master yet to match the firmware.

jgoppert, 2016-05-09:
It shows mag calibration done in messages. Recalibrating without rebooting does not help.

jgoppert, 2016-05-09:
Odd, on reboot it is set correctly without recalibrating, also I did a param refresh before I saved the params above, so something about the cal params not getting updated until reboot.

```bash
1	1	CAL_MAG0_ID	73225	6
1	1	CAL_MAG0_ROT	0	6
1	1	CAL_MAG0_XOFF	0.345971733331680298	9
1	1	CAL_MAG0_XSCALE	0.982476353645324707	9
1	1	CAL_MAG0_YOFF	-0.028831541538238525	9
1	1	CAL_MAG0_YSCALE	1.089622259140014648	9
1	1	CAL_MAG0_ZOFF	0.518945813179016113	9
1	1	CAL_MAG0_ZSCALE	1.050113677978515625	9
1	1	CAL_MAG1_ID	131594	6
1	1	CAL_MAG1_ROT	-1	6
1	1	CAL_MAG1_XOFF	0.053920418024063110	9
1	1	CAL_MAG1_XSCALE	1.000000000000000000	9
1	1	CAL_MAG1_YOFF	-0.187275499105453491	9
1	1	CAL_MAG1_YSCALE	1.000000000000000000	9
1	1	CAL_MAG1_ZOFF	-0.350996047258377075	9
1	1	CAL_MAG1_ZSCALE	1.000000000000000000	9
1	1	CAL_MAG2_ID	0	6
1	1	CAL_MAG2_ROT	-1	6
1	1	CAL_MAG_PRIME	73225	6
1	1	CAL_MAG_SIDES	63	6
```

jgoppert, 2016-05-15:
On the v4 firmware I didn't see this problem, so I will watch next time on v2 to see if it is fixed now.

dagar, 2016-11-08:
Still a problem?

LorenzMeier, 2016-12-26:
Fixed.

closed 2016-12-26 by LorenzMeier