PX4/PX4-Autopilot #4487: Battery current offset go lost during refactor

AndreasAntener, 2016-05-09:
@julianoes 
This offset is not considered anymore: https://github.com/PX4/Firmware/blob/master/src/modules/sensors/sensors.cpp#L322, please check the version before you pulled the battery out of the sensors app.

LorenzMeier, 2016-05-09:
Seems like that was a refactor from hell 8). If you run git blame you can see how many lines I already had to fix (and then to fix again because I committed more mistakes).

julianoes, 2016-05-09:
@AndreasAntener Sorry, I'll chase that up!

jgoppert, 2016-05-18:
Could this be related to https://github.com/PX4/Firmware/issues/4567?

AndreasAntener, 2016-05-18:
@jgoppert no this one is about a new setting for power modules that come with a fixed offset on the current signal.

jgoppert, 2016-05-18:
Any chance the acsp4 from auav that I have needs this param to correct the current to the extent that it would cause #4567?

AndreasAntener, 2016-05-18:
I don't know what settings they need, they don't specify it on their page so I'm assuming defaults should work? The offset I've only seen on one module so far (from Drotek)

LorenzMeier, 2016-05-22:
PR open for this.

closed 2016-05-22 by LorenzMeier