PX4/PX4-Autopilot #4489: Release testing summary

tumbili, 2016-05-09:
# Summary (250 quad with Pixracer connected over Wifi)

### Calibration step
#### Gyro calibration
Just say to hold the vehicle still!
#### Accel calibration
Last two pictures were cut off on Samsung tablet
#### Mag calibration
Mag orientation should be asked when clicking on the mag calibration tab. It shouldn't be just there on the side, the user could miss it.
### Flight modes
After setting the flight modes a message appeared that some parameters are missing (put screenshot here)
#### General stuff
- You always need to scroll down to see the latest messages
- Maybe some status message like "safe to fly" would be nice

### Flying
- On the first try takeoff only worked when putting it in the HOLD mode via tablet.
- On the second try takeoff command worked when vehicle was in MANUAL mode. But then the vehicle shot up and drifted off immediately as it was in MANUAL MODE (it held attitude). Do we really want this? It took me by surprise but maybe other users are much more aware that they need to take the vehicle over with the rc or switch it into a different mode. (put log here)
- Maybe a 3-2-1 counter could be added to not scare the shit out of the user
- LANDING: All in all works good, could be a bit faster killing the motors. Comes down, touches the ground and kind off hovers there (almost goes up again a bit) and then all of a sudden kills the motors. Nothing serious there.
- LANDING: Should we reset the landing mode after the vehicle was disarmed?
- Changing altitude via 'change altitude button' on tablet: We should allow negative values since it's relative to the current position (you can write negative values but not via the slider)
- Changing altitude via 'change altitude button' on tablet: Maybe the default value when pressing the button could be set to zero or something low. 100m is mostly far off what the user wants.
- Changing altitude via 'change altitude button' on tablet: Maybe we could also just save the previous value then the user has the chance to apply the same change over and over again.
- Battery voltage reading (setup 3S 2200mAh, AUAV power module, Pixracer): During flight the voltage would show 22% and after landing the voltage shows 58% (just and example)
- after uploading a mission the user lacks feedback of weather the upload was successful or not
- IMPORTANT: When flying a mission during land mode I tried to put the vehicle in hold mode. That lead to the vehicle falling down by about 5 meters and after that recovering back to the alt it should have been. Almost felt like throttle integrator was to zero. (Put log here)

kd0aij, 2016-05-09:
Did you have a map display in mission planning mode? I've had no success with offline maps.

tumbili, 2016-05-09:
@kd0aij Hm, good question. I think I had one but only at a certain zoom.

tumbili, 2016-05-09:
@DonLakeFlyer Maybe some of this is interesting for you.

dagar, 2016-05-09:
I like the countdown idea.

LorenzMeier, 2016-05-09:
Takeoff handling should now be fine. You need to update QGC and stable. We still need to fix the altitude handling in land mode.

LorenzMeier, 2016-05-11:
> On the first try takeoff only worked when putting it in the HOLD mode via tablet.

Fixed on master.

> On the second try takeoff command worked when vehicle was in MANUAL mode. But then the vehicle shot up and drifted off immediately as it was in MANUAL MODE (it held attitude, EDIT, it held altitude). Do we really want this? It took me by surprise but maybe other users are much more aware that they need to take the vehicle over with the rc or switch it into a different mode. (put log here) EDIT: Logs showed that I was having the rc throttle stick at mid throttle when I armed. That is why the vehicle took off

Arming via tablet now requires idle throttle

> Maybe a 3-2-1 counter could be added to not scare the shit out of the user

Will do, next release of QGC.

> LANDING: All in all works good, could be a bit faster killing the motors. Comes down, touches the ground and kind off hovers there (almost goes up again a bit) and then all of a sudden kills the motors. Nothing serious there.

Due to safety considerations: Will not change any of that right now.

> LANDING: Should we reset the landing mode after the vehicle was disarmed?

Yes, but not sure what you mean?

> Changing altitude via 'change altitude button' on tablet: We should allow negative values since it's relative to the current position (you can write negative values but not via the slider)

Added in QGC

> Changing altitude via 'change altitude button' on tablet: Maybe the default value when pressing the button could be set to zero or something low. 100m is mostly far off what the user wants.

Altitude change now limited to 15m, takeoff to 60m

> Changing altitude via 'change altitude button' on tablet: Maybe we could also just save the previous value then the user has the chance to apply the same change over and over again.

Yeah, next release. Right now smaller defaults will allow the user to make incremental adjustments.

> Battery voltage reading (setup 3S 2200mAh, AUAV power module, Pixracer): During flight the voltage would show 22% and after landing the voltage shows 58% (just and example)

Need to see the log

> after uploading a mission the user lacks feedback of whether the upload was successful or not

If you don't get an error message there is no error. We need to get away from positive confirmation. No news = success, as for any other modern user-facing software.

> IMPORTANT: When flying a mission during land mode I tried to put the vehicle in hold mode. That lead to the vehicle falling down by about 5 meters and after that recovering back to the alt it should have been. Almost felt like throttle integrator was to zero. (Put log here) See #4491

Fixed.

LorenzMeier, 2016-05-11:
Closing this as the bulk of this got addressed, in particular the safety related items.

closed 2016-05-11 by LorenzMeier