PX4/PX4-Autopilot #4490: Local Takeoff

tuofeichen, 2016-05-09:
Hey guys,
I was wondering if a local takeoff based on local position estimate (LPE) is implemented any where in the Firmware. I can send setpoint when the drone is on the ground, but at that point only a valid z estimate is present (since we're using optical flow moment). I will still try it later today, but just wondering if this feature is already implemented somewhere in the firmware. 

Thanks
Francis

tuofeichen, 2016-05-13:
nvm, can achieve on mavros side using setpoint

closed 2016-05-13 by tuofeichen

vinhk, 2016-07-15:
@tuofeichen But how can you detect x,y when there is likely that no data for x,y since optical flow is very close to the ground?

tuofeichen, 2016-07-15:
@vinhk , One hack is you can send velocity setpoint and let the drone jump off quickly, and as soon as the optical flow become valid, it will hover. 

I'm switching to visual SLAM at this point now, so the x, y is not constrained by the distance to the ground anymore.

vinhk, 2016-07-15:
@tuofeichen what topic did you use to make the velocity setpoint?

vinhk, 2016-07-15:
this I supposed ?
'/mavros/setpoint_velocity/cmd_vel'

vinhk, 2016-07-15:
`/mavros/local_position/velocity`

tuofeichen, 2016-07-15:
yeah, I think so, or whatever the most recent version of mavros is.

vinhk, 2016-07-15:
Is there a difference btween ` /mavros/setpoint_velocity/cmd_vel ` and 

` /mavros/local_position/velocity `

tuofeichen, 2016-07-15:
yeah, setpoint_velocity publish the setpoint* to* the pixhawk.
local_position is the LPE or whatever estimator you're using's estimate of
the local position/velocity of the drone, so it comes *from* the px4.

vinhk, 2016-07-15:
I see. I like to implement your idea of aggressive takeoff to a certain attitude so that it does not drift before it get to read optical flow data.

vinhk, 2016-07-15:
So in order to make it takeoff aggressively, I should use the setpoint_velocity?
but it is a subscribed topic. 
[image]

tuofeichen, 2016-07-15:
yes that's correct

vinhk, 2016-07-15:
would you be willing to share your old code sir?

vinhk, 2016-07-15:
I would like to takeoff fast and then stop once height is achieved.

vinhk, 2016-07-18:
would you be able to help me with implementing this method of aggressive takeoff and then hover once data is received?

tuofeichen, 2016-07-18:
Hey,
I doubt I have the implementation with me at this point, also I changed the
Firmware code a bit as well so I don't think you're gonna get it plug and
play. The tutorial on the dev website is pretty good, the structure at
least. with some slight modification you should be able to get it to work.
The key is always make sure your optical flow and rangefinder actually
gives you the right data. (suggest doing some handheld testing for the
position estimate before implementing offboard)

http://dev.px4.io/ros-mavros-offboard.html

vinhk, 2016-07-18:
could you help me with getting the data from the `mavros/px4flow/raw/optical_flow_rad` topic?
I am having trouble implementing a subscriber for it.

tuofeichen, 2016-07-20:
You need to plug in the flow module to your computer and download a px4flow
ros package to get that topic, the pixhawk connection doesn't provide that.
Also that's just for debugging purposes. You don't need to subscribe to
that topic, the i2c between flow and the pixhawk will provide position
estimate in the px4 firmware already.

vinhk, 2016-07-20:
I see. Yeah, I am thinking about maybe I can use the takeoff service where I can use the ultrasound sensor on the PX4Flow to get to the desire altitude and from there change the mode from OFFBOARD to POSCTL. Would this work?

tuofeichen, 2016-07-20:
no... takeoff service to my understanding needs GPS.

vinhk, 2016-07-20:
oh no. :(

vinhk, 2016-07-20:
I am wondering what did you use to have it takeoff vertically using Mavros?

tuofeichen, 2016-07-20:
you can send a position setpoint or velocity setpoint of z > 0.

vinhk, 2016-07-20:
I actually have a video of the testing using local position setpoint
https://www.youtube.com/watch?v=tApfAn-GJH4

vinhk, 2016-07-20:
velocity set point I am unsure of.

tuofeichen, 2016-07-20:
Most likely your position estimate is not correct. I suggest you log the
mavros/local_position and handheld the drone to maybe walk a known
trajectory and see if you get meaningful result.

vinhk, 2016-07-20:
by logging, I am unfamiliar with that. Do you mean print data using rostopic ?

mhkabir, 2016-07-20:
http://dev.px4.io/flight_log_analysis.html

The primary reason you keep running into issues is that you haven't studied the system basics and jumped straight into GPS-denied offboard control, which is a challenging proposition in itself. I would suggest that you first read up a bit on the system, and check out the PX4 devguide. You should also be prepared to jump into the code to achieve your goal.

If you're discussing something unrelated to the core flightstack bugs, I would suggest the PX4 Discuss : http://discuss.px4.io/

vinhk, 2016-07-20:
Okay, I'll do that.