PX4/PX4-Autopilot #4491: Hold mode during landing leads to the vehicle dropping down

tumbili, 2016-05-09:
When flying a mission during land mode I tried to put the vehicle in hold mode. That lead to the vehicle falling down by about 5 meters and after that recovering back to the alt it should have been. Almost felt like throttle integrator was to zero.

Explanation: The altitude setpoint during landing is way off the current altitude (because descend velocity is limited). Once we switch into hold the altitude setpoint is not reset but has to come up again first.
[image]

tumbili, 2016-05-09:
Log: http://logs.uaventure.com/view/s7Lnr8CYrvuRk887Qki876

LorenzMeier, 2016-05-09:
So entering hold should reset the alt SP immediately?

tumbili, 2016-05-09:
@LorenzMeier Yeah, I would say so.

tumbili, 2016-05-09:
@LorenzMeier I think this should be reproducible in SITL.

LorenzMeier, 2016-05-09:
@tumbili Can you look into this tomorrow?

closed 2016-05-10 by LorenzMeier

LorenzMeier, 2016-05-10:
@tumbili This is fixed on master. Please re-test outdoors, SITL looked good.

PR #4494 (2016-05-10): Navigator: Only run FOH logic when in mission mode
@tumbili This should fix #4491. Please test in SITL and real flights.