PX4/PX4-Autopilot #4492: RTK performance

tumbili, 2016-05-09:
Logs shows that RTK is not estimating well. During the flight the fix type switched back to 3 and form then on performance was great. GPS noise values are also interesting, they seem to drop after the switch back to fix type 3.
Log: http://logs.uaventure.com/view/ECA86gcfNagwdvSGYiXGkP
[image]

tumbili, 2016-05-09:
@bkueng @LorenzMeier

tumbili, 2016-05-09:
@bkueng @LorenzMeier FYI

mazgch, 2016-05-12:
@bkueng what is the RTCM_BUFFER_LENGTH in this test?

mazgch, 2016-05-12:
We should lower the UBX_TX_CFG_TMODE3_SVINACCLIMIT to get an accurate absolute position . Otherwise the absolute position error of the base translates directly to absolute position error of the UAV.

LorenzMeier, 2016-05-12:
@bkueng Can you look into these?

bkueng, 2016-05-12:
@mazgch RTCM_BUFFER_LENGTH was 250 bytes in this test. I'm not quite sure but I don't think we hit that boundary in this test. I never saw anything more than 180.
I extended it today to 300, but due to a limitation of mavlink, we internally need to fragment such long messages.

We don't check the CRC because this should be handled in the GPS device anyway. But if you want, you can create a pull request for it, and I'll merge it.

What do you suggest for UBX_TX_CFG_TMODE3_SVINACCLIMIT? I set it such that survey-in takes ~3-5min, which according to the spec should be ok. I think the relative position error is more important to us, at least for these tests.

tumbili, 2016-08-15:
Outdated

closed 2016-08-15 by tumbili